openrocket/openrocket #666: Error opening file with newest unstable

neilweinstock, 2020-05-11:
Opening attached ORK file gives the following error:
[image]

This cropped up recently; it works fine in the build from late April tagged 1c7171ef.

[blackfish_golden_sunset.ork.zip](https://github.com/openrocket/openrocket/files/4610684/blackfish_golden_sunset.ork.zip)

teyrana, 2020-05-13:
```
   > net.sf.openrocket.file.RocketLoadException: Exception loading stream: Index 2 out of bounds for length 2
   > 	at net.sf.openrocket.file.GeneralRocketLoader.load(GeneralRocketLoader.java:98)
   > 	at net.sf.openrocket.gui.util.OpenFileWorker.doInBackground(OpenFileWorker.java:72)
   > 	at net.sf.openrocket.gui.util.OpenFileWorker.doInBackground(OpenFileWorker.java:28)
   > 	at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304)
   > 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
   > 	at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343)
   > 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
   > 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
   > 	at java.base/java.lang.Thread.run(Thread.java:834)
   > Caused by: java.lang.IndexOutOfBoundsException: Index 2 out of bounds for length 2
   > 	at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:64)
   > 	at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:70)
   > 	at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:248)
   > 	at java.base/java.util.Objects.checkIndex(Objects.java:372)
   > 	at java.base/java.util.ArrayList.get(ArrayList.java:458)
   > 	at net.sf.openrocket.rocketcomponent.RocketComponent.getChild(RocketComponent.java:1498)
   > 	at net.sf.openrocket.rocketcomponent.SymmetricComponent.getPreviousSymmetricComponent(SymmetricComponent.java:587)
   > 	at net.sf.openrocket.rocketcomponent.Transition.getForeRadius(Transition.java:77)
   > 	at net.sf.openrocket.rocketcomponent.Transition.getRadius(Transition.java:356)
   > 	at net.sf.openrocket.rocketcomponent.FinSet.getFinFront(FinSet.java:938)
   > 	at net.sf.openrocket.rocketcomponent.FreeformFinSet.clampFirstPoint(FreeformFinSet.java:291)
   > 	at net.sf.openrocket.rocketcomponent.FreeformFinSet.update(FreeformFinSet.java:276)
   > 	at net.sf.openrocket.rocketcomponent.RocketComponent.componentChanged(RocketComponent.java:291)
   > 	at net.sf.openrocket.rocketcomponent.FinSet.componentChanged(FinSet.java:798)
   > 	at net.sf.openrocket.rocketcomponent.Rocket.fireComponentChangeEvent(Rocket.java:460)
   > 	at net.sf.openrocket.rocketcomponent.RocketComponent.fireComponentChangeEvent(RocketComponent.java:1761)
   > 	at net.sf.openrocket.rocketcomponent.RocketComponent.fireComponentChangeEvent(RocketComponent.java:1773)
   > 	at net.sf.openrocket.rocketcomponent.RocketComponent.fireAddRemoveEvent(RocketComponent.java:1485)
   > 	at net.sf.openrocket.rocketcomponent.RocketComponent.addChild(RocketComponent.java:1398)
   > 	at net.sf.openrocket.rocketcomponent.RocketComponent.addChild(RocketComponent.java:1353)
   > 	at net.sf.openrocket.file.openrocket.importt.ComponentHandler.openElement(ComponentHandler.java:52)
   > 	at net.sf.openrocket.file.simplesax.DelegatorHandler.startElement(DelegatorHandler.java:53)
   > 	at java.xml/com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser.startElement(AbstractSAXParser.java:510)
   > 	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLNSDocumentScannerImpl.scanStartElement(XMLNSDocumentScannerImpl.java:374)
   > 	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl$FragmentContentDriver.next(XMLDocumentFragmentScannerImpl.java:2710)
   > 	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLDocumentScannerImpl.next(XMLDocumentScannerImpl.java:605)
   > 	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLNSDocumentScannerImpl.next(XMLNSDocumentScannerImpl.java:112)
   > 	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanDocument(XMLDocumentFragmentScannerImpl.java:534)
   > 	at java.xml/com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:888)
   > 	at java.xml/com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:824)
   > 	at java.xml/com.sun.org.apache.xerces.internal.parsers.XMLParser.parse(XMLParser.java:141)
   > 	at java.xml/com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser.parse(AbstractSAXParser.java:1216)
   > 	at net.sf.openrocket.file.simplesax.SimpleSAX.readXML(SimpleSAX.java:49)
   > 	at net.sf.openrocket.file.openrocket.importt.OpenRocketLoader.loadFromStream(OpenRocketLoader.java:50)
   > 	at net.sf.openrocket.file.AbstractRocketLoader.load(AbstractRocketLoader.java:21)
   > 	at net.sf.openrocket.file.GeneralRocketLoader.loadUsing(GeneralRocketLoader.java:240)
   > 	at net.sf.openrocket.file.GeneralRocketLoader.loadRocket(GeneralRocketLoader.java:201)
   > 	at net.sf.openrocket.file.GeneralRocketLoader.loadStep1(GeneralRocketLoader.java:160)
   > 	at net.sf.openrocket.file.GeneralRocketLoader.load(GeneralRocketLoader.java:94)
   > 	... 8 more
```

teyrana, 2020-05-13:
.... 

```
 > Caused by: java.lang.IndexOutOfBoundsException: Index 2 out of bounds for length 2
   . . . 

   > at net.sf.openrocket.rocketcomponent.SymmetricComponent.getPreviousSymmetricComponent(SymmetricComponent.java:587)
   > 	at net.sf.openrocket.rocketcomponent.Transition.getForeRadius(Transition.java:77)
   > 	at net.sf.openrocket.rocketcomponent.Transition.getRadius(Transition.java:356)
   > 	at net.sf.openrocket.rocketcomponent.FinSet.getFinFront(FinSet.java:938)
```

teyrana, 2020-05-13:
#663  Is possibly at fault, due to inconsistent rocket state.  I need to talk a look.

teyrana, 2020-05-16:
Found the culprit.  switched `getStage()` calls to `getAssembly()` calls.

commit 93e34c47 (2020-05-17): Merge pull request #669 from teyrana/666-fix-oob

Resolves #666 -- fix the "Exception loading stream" error while loading certain rockets.
commit ce6a6fa2 (2020-05-17): [fixes #666] Find SymmetricComponent.PreviousSymmetricComponent now handles PodSets correctly.
closed 2020-05-17 by teyrana