openrocket/openrocket #667: OpenRocket 15.03 unable to launch through Java version 8 latest. Please help!!

ApoorvSom, 2020-05-12:
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00000000, pid=7560, tid=0x00002510
#
# JRE version: Java(TM) SE Runtime Environment (8.0_251-b08) (build 1.8.0_251-b08)
# Java VM: Java HotSpot(TM) Client VM (25.251-b08 mixed mode, sharing windows-x86 )
# Problematic frame:
# C  0x00000000
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

neilweinstock, 2020-05-13:
Apoorv got it working using a packaged installer.  Shall we close this?

teyrana, 2020-05-13:
well, thanks to whoever helped Apoorv?

teyrana, 2020-05-13:
Do we know what the inciting problem is?

neilweinstock, 2020-05-13:
He found my packaged installer, presumably from TRF.

Don’t know what the problem was. I just didn’t think we were going to be trying to chase down java problems on 15.03 at this point.

Ce message, ainsi que tous les fichiers joints à ce message, peuvent contenir des informations sensibles et/ ou confidentielles ne devant pas être divulguées. Si vous n'êtes pas le destinataire de ce message (ou que vous recevez ce message par erreur), nous vous remercions de le notifier immédiatement à son expéditeur, et de détruire ce message. Toute copie, divulgation, modification, utilisation ou diffusion, non autorisée, directe ou indirecte, de tout ou partie de ce message, est strictement interdite.

This e-mail, and any document attached hereby, may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized, direct or indirect, copying, disclosure, distribution or other use of the material or parts thereof is strictly forbidden.

teyrana, 2020-05-13:
Okay, well glad it got solved, then.

closed 2020-05-13 by teyrana

airqualityanthony, 2020-09-17:
What was the solution here? I'm having the same issue.