openrocket/openrocket #668: Exception while attempting to open a file

teyrana, 2020-05-13:
```
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NullPointerException
	at java.desktop/javax.swing.JComponent.repaint(JComponent.java:4851)
	at java.desktop/sun.swing.FilePane$3.repaintListSelection(FilePane.java:249)
	at java.desktop/sun.swing.FilePane$3.repaintSelection(FilePane.java:239)
	at java.desktop/sun.swing.FilePane$3.focusLost(FilePane.java:234)
	at java.desktop/java.awt.AWTEventMulticaster.focusLost(AWTEventMulticaster.java:238)
	at java.desktop/java.awt.Component.processFocusEvent(Component.java:6524)
	at java.desktop/java.awt.Component.processEvent(Component.java:6388)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5008)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
	at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1950)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:1059)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:725)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.SentEvent.dispatch(SentEvent.java:70)
	at java.desktop/java.awt.DefaultKeyboardFocusManager$DefaultKeyboardFocusManagerSentEvent.dispatch(DefaultKeyboardFocusManager.java:248)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.sendMessage(DefaultKeyboardFocusManager.java:275)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:795)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.SequencedEvent.dispatch(SequencedEvent.java:202)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
[... 375 lines cut ...]
 194     14.855 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
 195     14.855 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
 196     14.855 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
 197     14.855 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
 198     14.855 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.AccessController.doPrivileged(Native Method)
 199     14.855 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
 200     14.855 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
 201     14.855 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
 202     14.855 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
 203     14.856 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.AccessController.doPrivileged(Native Method)
 204     14.856 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
 205     14.856 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
 206     14.856 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
 207     14.856 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
 208     14.856 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
 209     14.856 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
 210     14.856 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
 211     14.856 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
 212     14.857 INFO   (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
 213     14.857 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
 214     19.870 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 215     19.880 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

```

teyrana, 2020-05-13:
(note: have not been able to reproduce since)

wolsen, 2020-05-14:
Looking at the stack trace, I feel this *may* be a java bug itself - not necessarily a bug within OpenRocket. Assuming that you're on JDK11 (which feels okay since the line numbers all match up nicely), the last call two calls in the stack are the following code:

JComponent.java
```java
    /**
     * Adds the specified region to the dirty region list if the component
     * is showing.  The component will be repainted after all of the
     * currently pending events have been dispatched.
     *
     * @param  r a <code>Rectangle</code> containing the dirty region
     * @see #isPaintingOrigin()
     * @see java.awt.Component#isShowing
     * @see RepaintManager#addDirtyRegion
     */
    public void repaint(Rectangle r) {
        repaint(0,r.x,r.y,r.width,r.height);
    }
```

FilePane.java
```java
        private void repaintListSelection(JList<?> list) {
            int[] indices = list.getSelectedIndices();
            for (int i : indices) {
                Rectangle bounds = list.getCellBounds(i, i);
                list.repaint(bounds); // <---- Line 249
            }
        }
```

The JList clearly indicates that there are some selected indices, otherwise the loop would be skipped. The `JList.getCellBounds(...)` function *can* return null, but this calling code does not expect it to ever be null. So it passes the bounds to the list.repaint(...) method which the freaks out with a NPE because the bounds rectangle is de-referenced in the repaint method.

Now, the interesting thing is the JList.getCellBounds() method. We can see that it can return null, but only when the getUI() method returns null.

JList.java
```java
    public Rectangle getCellBounds(int index0, int index1) {
        ListUI ui = getUI();
        return (ui != null) ? ui.getCellBounds(this, index0, index1) : null;
    }
```

The JList.getUI() method is not of much interest, since its simply an accessor to the ui parent class variable:

```java
    /**
     * Returns the {@code ListUI}, the look and feel object that
     * renders this component.
     *
     * @return the <code>ListUI</code> object that renders this component
     */
    public ListUI getUI() {
        return (ListUI)ui;
    }
```

What *is* interesting is that this all has to do with the L&F for the component and the environment itself. However, this is where I stop my code walking of the Java internals and suggest that it *might* be a java bug itself. There's clearly a race condition *somewhere* in which the JList.getCellBounds() method is invoked without the L&F being set for the component. I suppose that it is possible that OpenRocket code might be unintentionally causing this somewhere...

wolsen, 2020-05-14:
Also one thread I failed to mention aboved, but did pull on as well is that the call to `ui.getCellBounds(...)` within the JList.getCellBounds(int, int) method may also return null.

There's a couple of models behind it which have differing behaviors, but the javax.swing.plaf.basic.BasicListUI is the one that I investigated (the other is javax.swing.plaf.multi.MultiListUI and on the surface looks uninteresting for this case, but I didn't dig into as I believe the BasicListUI is all that's being used).

The salient points of this method are below, since index1 == index2.

BasicListUI.java
```java
    /**
     * {@inheritDoc}
     */
    public Rectangle getCellBounds(JList<?> list, int index1, int index2) {
        maybeUpdateLayoutState();

        int minIndex = Math.min(index1, index2);
        int maxIndex = Math.max(index1, index2);

        if (minIndex >= list.getModel().getSize()) {
            return null;
        }

        Rectangle minBounds = getCellBounds(list, minIndex);

        if (minBounds == null) {
            return null;
        }
        if (minIndex == maxIndex) {
            return minBounds;
        }
        ...
```

This method can return null if the index provided is greater than the list model's size. It can also return null if the `getCellBounds(list, minIndex)` invocation returns null (which it does, but only if the provided index > the model.size(), which it shouldn't be since that condition was checked previous to the call to `getCellBounds(list, minIndex)`.

I suspect that there's a race condition in there (there == Java codebase) somewhere that either:

1. fails to set the UI (the L&F) for the component before the repaint event occurs
2. An invocation is called to the repaintSelection while the component is still being setup
3. Changes the JList model's contents during this repaint sequence

Either way, its clear that the JList.repaintSelection() method does not expect null to ever return from the getCellBounds() method. The fact is that it _might_ return null (in fact, it did or we wouldn't have this lovely stacktrace) under some hypothetical and real circumstances. Regardless of whether there's code aggravating it in OpenRocket, there's clearly a bug in the underlying Java code base.

teyrana, 2020-05-15:
woah.  That's some detailed analysis!  
That's impressive :)

... but yea, I'm leaning towards just closing this issue as a one-off, unless we see it again.

teyrana, 2020-05-16:
closing : CNR

closed 2020-05-16 by teyrana