openrocket/openrocket #671: simulation results in rocket design pane don't match results in flight simulations tab

JoePfeiffer, 2020-05-18:
In the attached rocket, the Altitude in the rocket design pane is significantly different from the value shown in the Flight Simulations tab (Max velocity and Max acceleration are also different, though not by as much).

Rocket design pane:
[image]

Simulation tab:
[image]

[Rocket Double Stage Constant Diameter.ork.zip](https://github.com/openrocket/openrocket/files/4641514/Rocket.Double.Stage.Constant.Diameter.ork.zip)

neilweinstock, 2020-07-19:
Is this one fixed?  Should it be closed?

JoePfeiffer, 2020-07-19:
Not fully -- I got distracted, but am back on it.

closed 2020-11-01 by teyrana