openrocket/openrocket #679: Fin Moment of Inertia calcs.

KavanaghJake, 2020-06-12:
On line 712 of the finset file, the rotational MOI is calculated using a rectangular approximation for the fin, and the parallel axis theorem is included, as well as a multiplication by the number of fins. The mass is excluded, and then later included, in line 324 of the massCalculation file. The mass is referenced from the compCM.weight, I believe that when the compCM.weight is referenced, it uses the mass of all the fins in the set, put together. Since the MOI calculation has already been done with the 3 fins considered, I believe that value should only be multiplied by the mass for one fin. 

I took some steps to check this myself to be sure,

I rebuilt the identical rocket in solidworks, as well as calculated the MOI for the rocket by hand. when the fins are excluded, my hand calculations, solidworks, and the MOI that OpenRocket gives all agree. Then I added 1 fin, and the once again, still all agreed. When I have 3 fins, the rotational inertia that is contributed by the fins in openrocket is about 3 times what solidworks gives, and what my hand calculations give. My last step to check this, was I clicked 'split fins', creating individual instances of each fin as opposed to a set. When the split fin option is selected, the rotation moment of inertia agrees with solidworks, and is different than what I get when I have all the fins together as one component.

teyrana, 2020-06-12:
please clarify -- are you referencing the current commits of HEAD, or the last release, 15.03? 

Also, if you included the breakdown of the weights in this issue, it would help us to update suitable unit tests. :)   *particularly* if you can provide the .ork file.

KavanaghJake, 2020-06-12:
I am on the latest release, 15.03.

So to summarize I guess, the Moment of inertia calculations for fins are n times too large, where n is the number of fins in the fin set. When I split the fins, this fixes the issue. I quickly check this by just exporting the rotational moment of inertia with the fins together as a set (this is how they should be right now in the .ork file I will include), and then once again exporting the rotational moment of inertia with the fins split.

[△EXTERNAL]

teyrana, 2020-06-13:
@KavanaghJake 

Yes, I understand what you're saying. 

An existing set of validation data, including all the specifics would just make the process go quicker.

Also, the mass-calculation code has undergone significant change since 15.03, so we really need to (re)validate the data set itself, since the code has already change _anyway_, and may or may not include the same issue.

We take mass-calculation issues seriously, and I do want to fix them -- but we want to be sure we're  making the right change, first.

KavanaghJake, 2020-06-13:
I tried to reply in with an email to include to .ork file, but I am seeing now that it may not have been included. Is there an alternative way to attach the file? Github wont let me attach the .ork file here. I will include the rotational moment of inertia files here. I am using a 0 mass motor as that allowed me to compare between OpenRocket and Solidworks
(https://github.com/openrocket/openrocket/files/4773705/FinSetRotationalMOI.xlsx)
[SplitFinRotationalMOI.xlsx](https://github.com/openrocket/openrocket/files/4773706/SplitFinRotationalMOI.xlsx)

JoePfeiffer, 2020-06-13:
That's an annoyance with github that, as far as I know, can't be fixed.  We typicallyrename the file from (for instance) rocket-bad-moi.ork to rocket-bad-moi.ork.zip so github will let us upload it.

KavanaghJake, 2020-06-13:
[SimpleRocketDesignOpenRocket.zip](https://github.com/openrocket/openrocket/files/4773723/SimpleRocketDesignOpenRocket.zip)

I hope this worked!

JoePfeiffer, 2020-06-13:
That got it!  All that's really necessary is to change the file name -- .ork files are already zip-compressed.  But I was able to unzip it and get the rocket back out.  Thanks.  Now we see if the version in unstable is still doing it....

teyrana, 2020-06-21:
@JoePfeiffer 
Any progress on this? 
Do you have this handled, or should I take a look?

JoePfeiffer, 2020-06-22:
I've got it -- getting back to it.

JoePfeiffer, 2020-06-23:
> I tried to reply in with an email to include to .ork file, but I am seeing now that it may not have been included. Is there an alternative way to attach the file? Github wont let me attach the .ork [...]

Is that the correct Excel file?  All that seems to be in it is a couple of comments, then two columns of data -- the first going from 0 to 0.24 by .01, the second a constant 1.89e-05

KavanaghJake, 2020-06-23:
So what I did was just took one of the motors that has 0 mass in order to run a simulation where the MOI wouldn't be changing. so the 0.01 incremental data column is just the time for that sim to run, and then the second columns are the MOI. The first excel file, I had the fins together as a set. the second excel file, all I did was click 'split fins' and then run the sim again with the same 0 mass motor. I still had three fins on the rocket, they were just separated.

With the .ork file I attached, you should be able to do the exact same thing I believe. run it with the fins together, export the MOI's, and then run it with the fins split, and then once again export the MOI's

I also have done some thinking, and what should fix this, is if you go to line 712 of the finset file, delete the multiplication by the number of fins. Since the Mass is included later on in the calculation, and it is included for the entire finset as opposed to just the mass of one fin, by using the mass of all the fins in the fin set, it is essentially taking the MOI contribution for one fin, and multiplying it by the number of fins, twice.

line 712 of the finset calculates the MOI for one fin, excluding the mass, and then multiplies by the number of fins. in order to include the mass, it would be correct to multiply by the mass of one fin, since the number of fins has already been accounted for
line 324 of the mass calculation file multiplies by the mass of the entire fin set. if the mass calculation file were to multiply by the mass of one fin, this would also work, but I am guessing the mass calculation is used for other components so wouldn't want to change that. 
in line 712, if you were to take out the multiplication by the number of fins, this would work, because when you multiply by the mass of the entire finset in line 324, that is also factoring in the number of fins

I hope this makes sense, I am not sure if I am explaining it simply enough (but I hope I am lol). I am not sure if deleting the multiplication by the number of fins just for that one equation would cause any other issues, as I haven't really looked into that

JoePfeiffer, 2020-06-23:
I believe you are correct, and removing the multiplication by the number of fins would also make the calculation more similar to some of the aerodynamic calculation code.  I was just surprised to see an Excel file with all identical rows!

The problem still seems to exist in the unstable branch, and in addition another bug seems to have snuck in to getRotationalUnitInertia(), in which the shift of the inertia center to the fin set center seems to now happen regardless of whether there are multiple fins...  that's what I'm actually looking at now...

KavanaghJake, 2020-06-23:
Yeah sorry haha, I probably should have made that data much more informative + mentioned the 0 mass motor.

teyrana, 2020-06-24:
Like mot of the bugs in this release, the hard part comes down to multiple fins on pods or boosters -- 

We should definitely get the same results with grouped vs split fins, but it's hard to "relocate" MOIs because it's path dependent. -- the Parallel Axis Theorem is path dependent, so we have to calculate the entire configuration's C.M. and then calculate a single move.   Moving it in multiple steps results in a different answer. 

(if we could move it in multiple steps, that'd definitely be easier! )

What that means, is that I implemented (or intended to implement) the MOI calculations to calculate the MOI around each instance (i.e. each fin). And then move that MOI to each fin instance.   Is there an easier way to do it?  I sure wish there were....

PR #688 (2020-07-11): fixes Fin MOI calculations 
Fixes #679 by removing extra multiplication by number of fins, updates unit tests to match
closed 2020-07-11 by JoePfeiffer