openrocket/openrocket #683: Most stable build with booster pods

ChristianMichael11325, 2020-06-15:
This is not so much an issue as it is a question.

What was the first/most stable build that implemented booster pods? I’ve been having issues getting the most recent experimental build to work and I only need the booster pod feature compared to the stable build of OR.

wolsen, 2020-07-13:
Hi @ChristianMichael11325 can you clarify on what kind of issues you've been getting in the most recent experimental builds? I don't have a good answer for what was the most stable build that implemented booster pods, but I'd be curious to know what your current issues are.

teyrana, 2020-07-16:
there is no official support for pods yet.  The features _do_  exist in the development branch.  Unfortunately, the consensus is that code is still too buggy for even a beta / release candidate.