openrocket/openrocket #689: Feature request: easier way to add mass object after sustainer is overridden

neilweinstock, 2020-07-03:
This is a fairly common situation: After building the rocket, I measure mass and CG and override them for the whole sustainer.  Now I want to figure out how much nose weight I need to add, and I have a problem: I can't see the effect if I add it to the sustainer, because the sustainer is overridden at the highest level.

There are different ways to get around this, but all of them are hacks.  It would be nice if we provided an easy and straightforward way to do this sort of thing.

I'm not sure at the moment how it should work.

teyrana, 2020-09-03:
From this description, OR is doing exactly what you tell it.   

If you tell it to override it's mass calculations, it will. 

If you don't want it to, (because you're adjusting specific components _within_ the stage) then don't override the stage, and it will again calculate mass on a per-component basis -- including your Mass Object.

neilweinstock, 2020-09-03:
No, OR does *not* do what I am asking for.  Let me try again.

A common thing to do is to measure my finished rocket (mass and CG), and then determine how much nose weight I need.  I might even want to adjust the nose weight for different motors.

If I override the stage, then I have no easy way to adjust nose weight.

If I *don't* override the stage, then I have no easy way to actually set the OR model to match the mass and CG of my built rocket.

What I want is to override the stage to match my measured values, and then still have the ability to (somehow) add nose weight and see the effects.  This would require (somehow) a mass object that is *not* overridden by the stage overrides.  As I said in my first message, I'm not sure what's the best way to do this.  Needs further thought.  But it would be a nice little feature.

hcraigmiller, 2020-09-06:
A way to add the feature  could simply be to:

1. Change the option appearing on the Mass Component, Override tab from "Override mass and CG of all subcomponents " to "Do not override the mass and CG of this component." At this time, mass components cannot have subcomponents, so the option to override subcomponents is non-existent anyway. However, if this option needs to be preserved, then

2. Add an additional option to the Mass Component, Override tab. Below the existing "Override mass and CG of all subcomponents " option, the option "Do not override the mass and CG of this component and subcomponents" could be added.

In either case, the default would be off, but, if turned on, this component would not be overridden by either a parental override or a stage override.

My thoughts.

neilweinstock, 2023-08-23:
Here on TRF is yet another example of this issue coming up: https://www.rocketryforum.com/threads/openrocket-override-stage-mass-cg-never-launch-motor-for-variable-nosecone-weight.181692/

Here is a proposal for how to implement this.  Essentially, I agree with what Craig wrote above: simply provide an option in the override tab to exclude this component from parent overrides.  We can quibble over what the exact wording should be later.
[image]

When this is done, the component is highlighted in red in the component tree, because it really needs to stand out as "different".  Oh and of course no parent override icon(s) are shown for this component.

I am inclined to include this option for every component that can be overridden, possibly with special-case exceptions, provided that it is possible.  Or perhaps it should be limited to internal components only, that might make more sense.  I also don't know if it needs to be limited to mass override or works for CG and others as well (if we limit this to internal and mass components, then only mass and CG are relevant.)

SiboVG, 2023-08-23:
> [image]

I have two concerns with this proposal:
1. I don't like the red coloring. Sure, it needs to stand out, but red is really an indication that something is wrong with a component. I'd prefer a yellow-orange color.
2. I'm not sure if we want the "Exclude from parent overrides" to exclude all the overrides (mass, CG and CD). Would it be better to have a separate option for each override? CG and mass will probably always be paired together (either include both in the parent override, or exclude them), but I can imagine that you'd sometimes want the CD to be controlled entirely by the parent, but the mass/CG by the component itself.

[image]

JoePfeiffer, 2023-08-23:
I agree a compelling case has been made for the feature.

1. I like the way you've got the override tab set up.  I'd suggest the Exclude text should only be bold if it's checked.

2. I don't like highlighting the whole component the way it's shown in the tree.  I actually think we could just use the blue override icon and it would stand out well enough in a sea of grey; if that's not then picking a unique color (the red orange you're showing would be good) for the icon should highlight it OK.

3. We need to continue to think about what components it can be applied to.  The one component it definitely should is, of course, mass objects.  In order to avoid a component tree with various subtrees overridden and others not, I would say it probably should not be available for any component that can have children.  It's a lot easier to add features than to take them away, so maybe start by only having it available for mass components and see if we get users with a strong interest in something else?

neilweinstock, 2023-08-23:
1) I'm not married to the red color. Just something that stands out in the component tree.  Something in the yellow-orange family is OK but I would make it darker; it's not very contrasty against the white background in the sample image.

2) I'm not sure how I would apply the "blue override icon" as Joe suggested.  A mass object does not (usually) use an override at all, so it has no override icon associated with it.

3) Upon further thought I agree with Joe that limiting this to mass objects would be a good first step.  I think it covers a pretty high percentage of the most likely use cases and avoids some squirrelly situations.

4) Given #2, I don't think separate options are needed for each type of override.  I'm not convinced that separate overrides for Cg and Mass make sense.  Cd doesn't apply at all to mass components, so whatever settings we have don't matter here (we should probably remove that for non-aerodynamic components. but that's a separate issue).

5) As for the wording of the option: another approach is to invert it: make it something like "Include this component in parent overrides", and have it checked by default (probably internal representation would be the opposite, so it would default correctly if not present)

JoePfeiffer, 2023-08-23:
Checking....  you're right, it doesn't show an override icon.  I'd argue it should; anything that has a user-set mass should.

Thinking further about the separate CG and mass exclusion, I think you're right.  One and not the other  doesn't really make sense.

I don't quite no why phrasing it as "exclude" rather than "include" seems much more intuitive to me, but it does.