openrocket/openrocket #692: Bug report of OR in Dell machine with Windows 10

LuisaMirelle, 2020-07-07:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IllegalArgumentException: MALFORMED
	at java.util.zip.ZipCoder.toString(ZipCoder.java:58)
	at java.util.zip.ZipInputStream.readLOC(ZipInputStream.java:300)
	at java.util.zip.ZipInputStream.getNextEntry(ZipInputStream.java:122)
	at net.sf.openrocket.file.motor.ZipFileMotorLoader.load(ZipFileMotorLoader.java:56)
	at net.sf.openrocket.file.motor.GeneralMotorLoader.load(GeneralMotorLoader.java:36)
	at net.sf.openrocket.database.MotorDatabaseLoader.loadDirectory(MotorDatabaseLoader.java:124)
	at net.sf.openrocket.database.MotorDatabaseLoader.loadDatabase(MotorDatabaseLoader.java:69)
	at net.sf.openrocket.database.AsynchronousDatabaseLoader.doLoad(AsynchronousDatabaseLoader.java:103)
	at net.sf.openrocket.database.AsynchronousDatabaseLoader.access$100(AsynchronousDatabaseLoader.java:8)
	at net.sf.openrocket.database.AsynchronousDatabaseLoader$LoadingThread.run(AsynchronousDatabaseLoader.java:126)

---------- Thread information ----------
Thread[DatabaseLoadingThread,1,main]

---------- System information ----------
OpenRocket version: 15.03
OpenRocket source: default
OpenRocket location: C:\Users\luisa\AppData\Local\OpenRocket\app\OpenRocket-15.03.jar
JOGL version: 2.1.5
Current default locale: en
System properties:
  app.preferences.id=net/sf/openrocket/startup
  awt.toolkit=sun.awt.windows.WToolkit
  file.encoding=Cp1252
  file.encoding.pkg=sun.io
  file.separator=\
  java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
  java.awt.printerjob=sun.awt.windows.WPrinterJob
  java.class.path=C:\Users\luisa\AppData\Local\OpenRocket\app\OpenRocket-15.03.jar
  java.class.version=52.0
  java.endorsed.dirs=C:\Users\luisa\AppData\Local\OpenRocket\runtime\lib\endorsed
  java.ext.dirs=C:\Users\luisa\AppData\Local\OpenRocket\runtime\lib\ext;C:\WINDOWS\Sun\Java\lib\ext
  java.home=C:\Users\luisa\AppData\Local\OpenRocket\runtime
  java.io.tmpdir=C:\Users\luisa\AppData\Local\Temp\
  java.launcher.path=C:\Users\luisa\AppData\Local\OpenRocket
  java.library.path=C:\Users\luisa\AppData\Local\OpenRocket\app;C:\Users\luisa\AppData\Local\OpenRocket
  java.runtime.name=Java(TM) SE Runtime Environment
  java.runtime.version=1.8.0_111-b14
  java.specification.name=Java Platform API Specification
  java.specification.vendor=Oracle Corporation
  java.specification.version=1.8
  java.vendor=Oracle Corporation
  java.vendor.url=http://java.oracle.com/
  java.vendor.url.bug=http://bugreport.sun.com/bugreport/
  java.version=1.8.0_111
  java.vm.info=mixed mode
  java.vm.name=Java HotSpot(TM) 64-Bit Server VM
  java.vm.specification.name=Java Virtual Machine Specification
  java.vm.specification.vendor=Oracle Corporation
  java.vm.specification.version=1.8
  java.vm.vendor=Oracle Corporation
  java.vm.version=25.111-b14
  line.separator=\u000d\u000a
  os.arch=amd64
  os.name=Windows 10
  os.version=10.0
  path.separator=;
  sun.arch.data.model=64
  sun.awt.enableExtraMouseButtons=true
  sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler
  sun.boot.class.path=C:\Users\luisa\AppData\Local\OpenRocket\runtime\lib\resources.jar;C:\Users\luisa\AppData\Local\OpenRocket\runtime\lib\rt.jar;C:\Users\luisa\AppData\Local\OpenRocket\runtime\lib\sunrsasign.jar;C:\Users\luisa\AppData\Local\OpenRocket\runtime\lib\jsse.jar;C:\Users\luisa\AppData\Local\OpenRocket\runtime\lib\jce.jar;C:\Users\luisa\AppData\Local\OpenRocket\runtime\lib\charsets.jar;C:\Users\luisa\AppData\Local\OpenRocket\runtime\lib\jfr.jar;C:\Users\luisa\AppData\Local\OpenRocket\runtime\classes
[... 95 lines cut ...]
 275    230.292 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2210
 276    231.301 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2220
 277    232.314 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2230
 278    233.324 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2240
 279    234.335 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2250
 280    235.344 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2260
 281    236.360 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2270
 282    237.375 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2280
 283    238.383 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2290
 284    239.392 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2300
 285    240.402 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2310
 286    241.415 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2320
 287    242.424 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2330
 288    243.436 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2340
 289    244.464 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2350
 290    245.474 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2360
 291    246.544 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2370
 292    247.569 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2380
 293    248.580 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2390
 294    249.590 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2400
 295    250.652 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2410
 296    251.660 DEBUG  (BlockingMotorDatabaseProvider.java:79) Database not loaded, count=2420
 297    251.945 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 298    251.949 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

wolsen, 2020-07-07:
It looks like the error is occurring when loading local ThrustCurve files. I can't tell which file in particular is causing the problems. Can you try moving the local ThrustCurve files to a different location and starting up OpenRocket to help confirm this is the case?

wolsen, 2020-07-08:
From rocketryforum discussion:

@LuisaMirelle , I was looking at your bug report and it looks as though the error is occurring when loading your local motor database files. The problem appears to happen when reading in some of the local thrustcurves. So a couple of questions for you:
To confirm, do you have local thrustcurve files? I'm not a Windows guy, but I believe it will be in one of the following for your system. You can also just search for the ThrustCurves folder:
C:\Users\luisa\OpenRocket\ThrustCurves
C:\Users\luisa\AppData\Local\OpenRocket\ThrustCurves
Can you move the local thrustcurve files to another location and try to start up OpenRocket?
Have you recently added new thrustcurve files? I can't tell which one in particular is causing the problem
I'll also comment on the bug report.

wolsen, 2020-07-13:
User seems to have resolved the issue [per the forum](https://www.rocketryforum.com/threads/openrocket-packaged-installers-for-windows-and-mac-to-solve-all-your-java-problems.143540/post-2019518). Closing this for now. Please re-open if you encounter this again.

closed 2020-07-13 by wolsen