openrocket/openrocket #693: Uncaught except. on Mac OS X 10.13.6 - auto error reporting failed with 404 instead of 202

robertrau, 2020-07-10:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.RuntimeException: Error starting OpenRocket
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:40)
	at net.sf.openrocket.startup.OpenRocket.main(OpenRocket.java:32)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:38)
	... 1 more
Caused by: java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1367)
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1342)
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1480)
	at net.sf.openrocket.startup.SwingStartup.main(SwingStartup.java:76)
	... 6 more
Caused by: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
	at com.google.inject.internal.util.$CustomConcurrentHashMap$ComputingImpl.get(CustomConcurrentHashMap.java:2041)
	at com.google.inject.internal.FailableCache.get(FailableCache.java:50)
	at com.google.inject.internal.ConstructorInjectorStore.get(ConstructorInjectorStore.java:49)
	at com.google.inject.internal.ConstructorBindingImpl.initialize(ConstructorBindingImpl.java:125)
	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:521)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
	at com.google.inject.internal.FactoryProxy.notify(FactoryProxy.java:46)
	at com.google.inject.internal.ProcessedBindingData.runCreationListeners(ProcessedBindingData.java:50)
	at com.google.inject.internal.InternalInjectorCreator.initializeStatically(InternalInjectorCreator.java:133)
	at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:106)
	at com.google.inject.Guice.createInjector(Guice.java:95)
	at com.google.inject.Guice.createInjector(Guice.java:72)
	at com.google.inject.Guice.createInjector(Guice.java:62)
	at net.sf.openrocket.startup.SwingStartup.runInEDT(SwingStartup.java:143)
	at net.sf.openrocket.startup.SwingStartup.access$000(SwingStartup.java:43)
	at net.sf.openrocket.startup.SwingStartup$1.run(SwingStartup.java:79)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:306)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
	at com.google.inject.internal.util.$CustomConcurrentHashMap$ComputingImpl.get(CustomConcurrentHashMap.java:2041)
	at com.google.inject.internal.FailableCache.get(FailableCache.java:50)
	at com.google.inject.internal.MembersInjectorStore.get(MembersInjectorStore.java:65)
	at com.google.inject.internal.ConstructorInjectorStore.createConstructor(ConstructorInjectorStore.java:73)
	at com.google.inject.internal.ConstructorInjectorStore.access$000(ConstructorInjectorStore.java:28)
	at com.google.inject.internal.ConstructorInjectorStore$1.create(ConstructorInjectorStore.java:36)
	at com.google.inject.internal.ConstructorInjectorStore$1.create(ConstructorInjectorStore.java:32)
[... 240 lines cut ...]
  84      3.890 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:529)
  85      3.890 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
  86      3.890 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
  87      3.890 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
  88      3.890 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
  89      3.891 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
  90      3.891 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.SingleFieldInjector.<init>(SingleFieldInjector.java:41)
  91      3.891 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.getInjectors(MembersInjectorStore.java:125)
  92      3.891 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.createWithListeners(MembersInjectorStore.java:95)
  93      3.891 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore.access$000(MembersInjectorStore.java:34)
  94      3.891 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:42)
  95      3.891 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:39)
  96      3.891 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.FailableCache$1.apply(FailableCache.java:39)
  97      3.891 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:549)
  98      3.891 STDERR (SwingExceptionHandler.java:49) 	... 44 more
  99      3.891 STDERR (SwingExceptionHandler.java:49) Caused by: java.lang.ClassNotFoundException: javax.script.ScriptEngineFactory
 100      3.891 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:435)
 101      3.891 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:589)
 102      3.891 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
 103      3.891 STDERR (SwingExceptionHandler.java:49) 	... 66 more
 104      3.892 INFO   (SwingExceptionHandler.java:64) Exception handler not on EDT, invoking dialog on EDT
 105      3.892 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
 106     14.172 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 107     14.180 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2020-07-10:
Unfortunately, OR 15.03 does not run with Java versions more recent than 8 (according to your bug report you are running java.vm.version=14.0.1+7).  That's one of the many things being addressed in the current development version!  Your best bet is to use the packaged installer for OSX which is on dropbox at https://www.dropbox.com/s/33o53vlle5b0uex/OpenRocket-15.03.dmg?dl=0
Instructions are in a thread on The Rocketry Forum at https://www.rocketryforum.com/threads/openrocket-packaged-installers-for-windows-and-mac-to-solve-all-your-java-problems.143540/

robertrau, 2020-07-10:
Thank you for your help!

Bob

closed 2020-07-13 by JoePfeiffer