openrocket/openrocket #698: Photo Studio doesn't listen properly for design changes

neilweinstock, 2020-07-14:
This problem goes back at least to 15.03.

Photo Studio does not react consistently to design changes.  If I have the Photo Studio open while tweaking the design, then it is a crapshoot whether the tweaks will be reflected in the Photo Studio.  Decal (texture) updates are particular bad in this regard.

Generally, the only way to fix this is to close the Photo Studio and reopen it.  Then you get to update all the settings again (see #697).

I can provide some documentation of this if you like, but it's easy to demonstrate with any design.

hcraigmiller, 2021-06-14:
@neilweinstock: In Windows 10 with Java version 11.0.9 2020-10-20 LTS, design changes are given effect _when the Photo Studio image is moved_ ever so slightly (unless the image is moved, the design change does not take effect in Photo Studio). Does your system function the same... is this still an issue?

neilweinstock, 2021-06-14:
So far my behavior matches what you say... but I'm not ready to say that's a complete answer.  I could swear that in the past I moved the rocket position around and it still didn't pick up the design changes.  I'll need to play with it some more.

But nudging the rocket looks like it will cause things to synch up at least *some* of the time, which is an improvement.

neilweinstock, 2022-01-27:
This was fixed at some point; it is quite responsive now.  Closing...

closed 2022-01-27 by neilweinstock