openrocket/openrocket #700: Working with simulation listeners

niniack, 2020-07-16:
I was working with [this guide](http://wiki.openrocket.info/Simulation_Listeners) to get started with some extension I want to build using OpenRocket. 

However, [this thread](https://www.rocketryforum.com/threads/running-openrocket-from-eclipse.67463/) implies that the instructions there are are out of date. 

I followed the guide and at this step: 
```
java -cp OpenRocket-1.1.6.jar:. net.sf.openrocket.startup.Startup
````
I ran into the following erro: 
```
Error: Could not find or load main class net.sf.openrocket.startup.Startup
```
If someone could point me in the right direction in developing an extension, I'd be very grateful!

teyrana, 2020-07-16:
A couple comments
1. yea, that guide is probably out of date  (comments and/or pull requests welcome!)
2. The simulation listener code, is relatively stable, afaik.  So no guarantees, but it should work mostly the same.
3. When you're running this, are you running this from source?  Git?  which branch?
4, What Java version are you using? We recently made the jump from Java 8 to Java 11
5. Beyond that, running `java -jar ${PATH}/OpenRocket.jar` should be working, (although most developers are using Eclipse, ant, 
or Intellij.)

niniack, 2020-07-16:
I'm working with the 15.03 release downloaded from the [main page](http://openrocket.info/). This is the [link](https://github.com/openrocket/openrocket/releases/download/release-15.03/OpenRocket-15.03.jar) it leads to. 

I'm working with Java 8, because I couldn't get OpenRocket to run with Java 11.

Yeah the final command works (given Java 8), but I'm still having issues with the extension. I'm working with [this](https://github.com/plaa/ThrustScaler) repo now. Seem to be having some luck.

teyrana, 2020-07-16:
Ahhhh, okay.  Yea:
15.03 is only compatible with java 8.  
Subsequent releases move to later version of java (primarily 11 for 20.xx).  

Beyond that, I don't have much experience with this corner of things....

niniack, 2020-07-16:
Oh man, maybe I'm missing something incredibly obvious, but how do I go about getting versions beyond 15? Sorry if this is too simple an issue

JoePfeiffer, 2020-07-16:
15.03 remains the most recently released version.  You can download or clone the current development, or "unstable" branch from the Code tab, and build it to test.  It isn't ready for a real beta release yet, but it is perfectly suitable for day-to-day use.

closed 2020-07-18 by niniack