openrocket/openrocket #704: Graceful behavior when opening future file formats

neilweinstock, 2020-07-19:
15.03 does not behave gracefully when opening files from the current unstable.  The new release should have a better approach to opening files created by future versions.

Recommend that the new release recognize a future file format and put up a clear message to the user that the file format is not supported.  At this point you could either require a confirmation from the user to try to open the file anyway, or just go ahead and try automatically, I don't know.

Examples of bad behavior in 15.03 that we should fix:

1) Here it is opening a new file:
[image]
The message that it's an unsupported document format is mushed in with all the other warnings, rather than being displayed on its own in more obvious way.

2) In a file that has fins attached to a transition, I get this:
[image]
This time, the program doesn't even get to the point of notifying the user that it's an unsupported document format; it just bails out with that error, and no useful info for the user.

teyrana, 2020-11-07:
## Breadcrumbs

1. net.sf.openrocket.file.openrocket.importt.DocumentConfig.java:55:
`public static final String[] SUPPORTED_VERSIONS = . . . `

teyrana, 2020-11-07:
## Fixed

This is the result of opening a future-file-version on the current development branch:

[image]

This sounds like what you're asking for, yes?

hcraigmiller, 2020-11-07:
I would suggest including the Java version required as well to identify a potential problem such as that with the current build and 15.03 requiring different Java versions.

neilweinstock, 2020-11-07:
How is that different from what was there before?  Not clear to me from the picture.