openrocket/openrocket #706: new fillet material should default to "dry white glue"

teyrana, 2020-07-21:


mamessner1980, 2025-08-08:
Yes – it looks like someone took the time to make custom fillets available with adjustable radius, but the materials are things like “cardboard” “balsa wood.”

Having material choices like (cured) white glue, wood glue, epoxy, epoxy with cabosil, and epoxy putty would be fantastic. 

FWIW, I got these densities from ChatGPT:

Material
Approx. Density (g/cm³)
Notes

Cured White Glue (PVA)
1.1 – 1.2
Elmer’s is around 1.15

Cured Wood Glue (e.g., Titebond)
1.2 – 1.3
Slightly higher than white glue

Cured Epoxy (5-min or 30-min)
1.1 – 1.3
Depends on brand/resin/hardener mix

Epoxy Putty
1.7 – 2.2
Heavily filled; JB Weld is ~1.85

Epoxy + Cabosil (fumed silica)
0.8 – 1.1
Thixotropic, full of air pockets

neilweinstock, 2025-08-08:
I started a thread on TRF to discuss: https://www.rocketryforum.com/threads/density-of-various-epoxies-openrocket-question.193598/.

I'm not saying I don't trust the numbers ChatGPT reports, but.... well yeah I am saying that. :)   Nonetheless these are a very good starting point.