openrocket/openrocket #710: Two repeatable crashers working with transitions

neilweinstock, 2020-07-26:
This is repeatable on the last few builds I am working with here on MacOS.  I'm guessing they're related so including both here.  If they're not then this can be split into two issues.

To recreate, scenario 1:
1) Open the "Simple model rocket" example.
2) Select "Body Tube"
3) Add a transition
4) Click "Close" button on transition editing dialog.
5) Click the "Delete" button.
This reliably creates the crash in attached file "transition_crash_1.txt".
[transition_crash_1.txt](https://github.com/openrocket/openrocket/files/4978916/transition_crash_1.txt)

Scenario 2:
1-3) Repeat as above
4) Select the "Aft Diameter" field
5) Type in a new number, I usually just use "1"
6) Hit TAB
The reliably creates the crash in attached file "transition_crash_2.txt"
[transition_crash_2.txt](https://github.com/openrocket/openrocket/files/4978915/transition_crash_2.txt)

wolsen, 2020-07-27:
Hmm, these are both SIGABRT in Java threads (Java 14)

neilweinstock, 2020-07-27:
Yes indeed, it appears this is a Java 14 problem.  Does not reproduce on Java 11.

Do we even want to track this or should I just close it?

hcraigmiller, 2022-02-12:
Where we will now be looking at JAVA 17 LTS compatibility, I recommend closing this issue.

SiboVG, 2022-02-12:
This issue did not reproduce in Java 17, will therefore close this issue

closed 2022-02-12 by SiboVG