openrocket/openrocket #711: Repeatable crashers working with freeform fin set in this design

neilweinstock, 2020-07-26:
This design was created by another user in 15.03 I was going to add pods on for him in the new unstable to get him a CP value.  In the process I discovered repeatable crashers with the main finset.

Here's the design:
[zeerust_scratch2.ork.zip](https://github.com/openrocket/openrocket/files/4978946/zeerust_scratch2.ork.zip)

Two crash scenarios.  First:
1) Open zeerust_scratch2.ork design
2) Select "freeform fin set"
3) Click "Delete"

This produces the following crash log:
[fin_crash_1.txt](https://github.com/openrocket/openrocket/files/4978957/fin_crash_1.txt)

On exactly one occasion it *didn't* crash.  In that case, I hit ctrl-Z and *then* it crashed. 

Second scenario:
1) Open zeerust_scratch2.ork design
2) Drag "Freemform fin set" to just under the transition, above the mass object

This produces the following crash log:
[fin_crash_2.txt](https://github.com/openrocket/openrocket/files/4978959/fin_crash_2.txt)

neilweinstock, 2020-07-27:
Update: almost *anything* I do to this design crashes OR.  This includes:
1) deleting the tube fin set (usually, maybe not 100% of the time)
2) Almost any changes to the fins
3) Adding a pod set to the main body tube

neilweinstock, 2020-07-27:
This appears to be Java 14-specific, does not reproduce on Java 11.

hcraigmiller, 2022-02-12:
Where we will now be looking at JAVA 17 LTS compatibility, I recommend closing this issue.

SiboVG, 2022-02-12:
This issue did not reproduce in Java 17, will therefore close this issue

closed 2022-02-12 by SiboVG