openrocket/openrocket #712: Update motor database

neilweinstock, 2020-07-27:
The motor database should be updated for the new version.  This issue is to track that task.

JoePfeiffer, 2020-07-27:
Should be mentioned that while the task should be tracked, it's trivial -- running 'ant serialize-motors' from .../core downloads the entire current database from thrustcurve.org, and the next build of OR will include it in the .jar file.  The version in the current unstable branch was current as of May 1.

neilweinstock, 2020-07-27:
OK, so I guess this is just a double-check that the update should be run one last time before final release.  Didn't realize the unstable builds had the updated database already.

If this issue is not needed, feel free to close.

JoePfeiffer, 2020-07-27:
I thought about suggesting that, but decided it is something we need to make sure we check off the list so I think it should remain open.

bdale, 2021-02-26:
Note that since some of the motor data files on thrustcurve.org have Tripoli Motor Testing copyright assertions, that should be noted somewhere in the license docs.  Here's the text I've added to debian/copyright in my packaging from source to address this:

> Files: core/resources/datafiles/thrustcurves/thrustcurves.ser
Copyright: 1997-2001 Tripoli Motor Testing
License: Freely redistributable
 This file contains motor performance data obtained by upstream from 
 thrustcurve.org, a resource maintained by and for the hobby rocketry 
 community.
 .
 While those who author motor data files may assert copyright, very few do, 
 and any data submitted to thrustcurve.org is intended and understood to be 
 freely redistributable.

neilbalch, 2021-05-11:
> Should be mentioned that while the task should be tracked, it's trivial -- running 'ant serialize-motors' from .../core downloads the entire current database from thrustcurve.org, and the next build [...]

Would it make sense to have the ability to dynamically update this by allowing the user to perform the update in the OR UI?

hcraigmiller, 2021-05-11:
Yes, please…

neilweinstock, 2021-05-11:
Would be nice but it doesn't seem like the most urgent thing to do to get the release out.

Frankly, I'm a touch confused why the library is baked into the build right now the way it is.  You can update the entire library by grabbing files from Thrustcurve.org that each have all the motors from one manufacturer.  So right now I have in my motors folder a set of those files, and voila my library is up-to-date.  So my motor folder looks like this:
[image]
(those are the only three manufacturers I use, so I didn't bother to fetch the rest)

Of course, now I can't find where on Thrustcurve you get those files.

The point being, OR has an internal motor database, but it's also quite capable and happy to load the data from external files.  Why the redundant capabilities?  Why not just package in those exact .eng files from Thrustcurve.  There aren't that many motor manufacturers, so it's not that many files.  Then updating would simply be a matter of grabbing the latest from Thrustcurve and replacing the internal ones.

Perhaps this is not an important question, just been curious about it.

bdale, 2021-05-11:
Joe Pfeiffer ***@***.***> writes:

> Letting users do the update is a very good idea for a future release
> (assuming John doesn't mind).

From a UI design standpoint, even better might be to do something like
using the API to find out during motor selection whether the local data
is present and up to date, or if the user should be offered the option
to fetch the current data for motors of interest?

Bdale

jimmiedave, 2022-02-21:
Don't know if it's useful, but Jan/Feb '22 Sport Rocketry has a comprehensive CAR/NAR/TRA Rocket Motor Certification list, which goes to several magazine pages, and from 1/8A.2-1 to 025000 (which, tbh, sounds like it must be a stick of dynamite).

Not suggesting anyone key this in (though I could OCR it, if desired), but it might be a helpful sanity check against what's in the databases currently.

hcraigmiller, 2022-02-21:
> Don't know if it's useful, but Jan/Feb '22 Sport Rocketry has a comprehensive CAR/NAR/TRA Rocket Motor Certification list, which goes to several magazine pages, and from 1/8A.2-1 to 025000....

If you are referring to the Cesaroni 30795O25000-P, that motor has been added to ThrustCurve.org, and it appears in the current OpenRocket-Unstable motor list. As to other motors, if those motors are listed on ThrustCurve, then those motors will most likely be included in the OpenRocket motor database.

JoePfeiffer, 2022-02-21:
If it's at thrustcurve.org, it *will* be in the OR database -- one of the last things we're going to do before release is pull a new copy.

The only thrustcurve.org motors not in our database are ones that contain errors that make it impossible to reliably read them (for instance, a few motors have more than one thrust value at a single time point).

closed 2024-02-04 by neilweinstock