openrocket/openrocket #713: Update Parts Database

neilweinstock, 2020-07-27:
The parts database in OR should be updated for the new version.  This issue is to track that task.

Maybe just import and use Dave Cook's database?  https://github.com/dbcook/openrocket-database

wolsen, 2020-07-27:
Dave Cook's database is Apache licensed, which is not necessarily incompatible with the GPL that OpenRocket is licensed under, but specific parts would need to call out which licenses they carry.

neilweinstock, 2020-07-28:
Hmm, in a previous message to the OR mailing list Dave said he'd be willing to switch to a different license but not GPL.  Is it possible to embed a database with a different license like that?  I am woefully ignorant of the ins and outs of this.  It would, however, be a real shame for us not to be able to use this incredibly useful resource, which is being given to us for free, due to license incompatibility.

Here's what Dave wrote:
> It's currently on the Apache license.  If that causes any heartburn I'd be happy to add a choice for a substantially similar license that you like better.  I would probably not agree to GPL it though.

teyrana, 2020-07-29:
This is not a problem, we just need to add this to the documentation -- i.e. that the motor database is licensed under Apache, instead of GPL.  They're otherwise compatible.

neilweinstock, 2020-07-29:
If true that's great.

hcraigmiller, 2021-03-29:
Part of my conversation with @neilweinstock:

"Okay, I spoke with my friend, an intellectual property attorney, about this. And, he said the GPL licensed software can incorporate another more restrictive license, such as the Apache license, for the libraries. He and I discussed a variety of ways to do so, including having a license page on first use that must be accepted, and placing the licensing restrictions on the splash page and license and about tab pages. He would be willing to look at what is developed and give comments and suggestions as well."

Food for thought...

hcraigmiller, 2022-01-31:
Is Dave's parts database supposed to be included in the 01/31/2022 merge?

JoePfeiffer, 2022-02-02:
Database included by #1077 and #1104.  Leaving this open due to conversation about licensing issues (see also #1079)

hcraigmiller, 2022-03-22:
dbcook/openrocket-database was updated to day; the update needs to be added to Beta 2.

SiboVG, 2022-03-22:
> dbcook/openrocket-database was updated to day; the update needs to be added to Beta 2.

Updated through PR #1256, thanks for notifying us on this @hcraigmiller

closed 2024-02-04 by neilweinstock