openrocket/openrocket #718: A 0m launchrod is cleared whilst the rocket is still on the ground

Zentrik, 2020-07-27:
I am simulating an actively stabilised rocket, that doesn't use a launchrod so I set it to 0m. However, when the rocket's height is less than 1e-20 (including negative heights) the launchrod is set to cleared. 
I believe the problem is with this line
https://github.com/openrocket/openrocket/blob/9a0ef5f8edbd6018621606ed8ff0555fd29a2b39/core/src/net/sf/openrocket/simulation/BasicEventSimulationEngine.java#L185

I tried changing it to `relativePosition.z - currentStatus.getSimulationConditions().getLaunchRodLength() > 1e-10`, to deal specifically with the no launchrod case however, around the time that the rocket would have lifted off the simulation just hung.

`relativePosition.z` is used to account for negative heights, which seems like a bug in itself which could be fixed by adding`relativePosition = currentStatus.getRocketPosition().sub(origin);` after https://github.com/openrocket/openrocket/blob/9a0ef5f8edbd6018621606ed8ff0555fd29a2b39/core/src/net/sf/openrocket/simulation/BasicEventSimulationEngine.java#L165. And 1e-10 is there to account for relativePosition.z having a small value when it is supposed to be 0.

JoePfeiffer, 2020-07-27:
I'm missing how this is a problem -- you've set the launchrod length to 0, so the moment the rocket leaves the ground it's cleared it.

My bigger question is how you're simulating an actively stabilized rocket -- OR doesn't have any support for anything but fixed fins (and motor angle), so it has no support for active stabilization.

Zentrik, 2020-07-27:
Sorry if I wasn't clear, I am simulating the rocket that I am going to use for active stabilisation in OpenRocket without the actual active stabilisation. Also, the rocket has not left the ground when the launchrod is set as cleared, as `relativePosition.z` can be negative, i.e. the rocket is below the ground, or very small such as 1e-60 which i think should be taken to be 0.

JoePfeiffer, 2020-07-27:
I'm sure I'm being dense, but I'm still not seeing it.  length() is what you want; that's the distance the rocket has moved from the origin, which may indeed be greater than z (especially if you've got a significant launch rod angle).  The bottom of the launch rod is at the same origin as the bottom of the rocket.  When the rocket has moved a distance greater than the effective length of the launch rod, it is marked as having cleared the rod.

You mention z might be below the ground; it will still be positive in this case, since it's relative to the origin.  Incidentally, how did you get a negative relative height?

Let me try to approach this from another direction:  when would you expect the rocket to clear a launch rod of length 0?  And how is the fact that it is clearing it immediately affecting your simulation results?

Zentrik, 2020-07-27:
I would expect the rocket to clear a launch rod of length 0 when the rocket is no longer touching the ground, I believe this is equivalent to `relativePosition.z > 0`. The fact that it clears immediately means that the rocket is rotating whilst still on the ground and moving horizontally, which is unlikely to occur in real life.

I presume I get a negative relative height, as relative position is calculated before the height is checked for being less than 0, so if say height is -0.1, that will be changed to 0 whilst relative position stays at -0.1. I believe the height is negative as the acceleration would be downwards due to gravity as thrust is less than gravity.

When you have a 0m launch rod, the angle doesn't make a difference however, if the negative relative position problem is fixed, length() can be used. Or there could be a separate setting indicating the lack of a launch rod.

I'm not sure how you can have a positive z if the rocket is below the ground, as the origin's height would be at ground level.

JoePfeiffer, 2020-07-27:
OK, I can see how it could happen; should probably reset relativePosition to 0 when the rocket's position is reset to the origin; also probably only check for launch rod clearance if the rocket has lifted off.

I remain curious how this is affecting your simulation...

Zentrik, 2020-07-27:
Well as the rocket can rotate whilst its on the ground, it led to a difference of a couple of meters in the apogee. However, if the rocket was stable it could have been more significant.

PR #721 (2020-07-29): Assure rocket has left the ground before it is considered as having cleared launch rod
Fixes #718
Reset rocket's relative position when its absolute position is reset.
Make sure rocket has lifted off before setting launch rod as cleared.
closed 2020-07-29 by JoePfeiffer