openrocket/openrocket #719: Bug Report - Small 3d and Photo Studio display

hcraigmiller, 2020-07-27:
---------- Bug report ----------

Include detailed steps on how to trigger the bug:

1. Change view from 2d Side View to 3d Finished
2. Or, View in Photo Studio
3. 

What does the software do and what in your opinion should it do in the case described above:

Small 3d display, window is not full screen in Photo Studio.

Include your email address (optional; it helps if we can contact you in case we need additional information):

hcraigmiller@gmail.com

(Do not modify anything below this line.)
---------- System information ----------
OpenRocket version: 19-xx-alpha-12
OpenRocket source: default
OpenRocket location: C:\Users\Tracy E. Miller\Downloads\OpenRocket\OpenRocket.New\OpenRocket-701-restore-optimizer.jar
JOGL version: null
Current default locale: en_US
System properties:
  awt.toolkit=sun.awt.windows.WToolkit
  file.encoding=Cp1252
  file.separator=\
  java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
  java.awt.printerjob=sun.awt.windows.WPrinterJob
  java.class.path=C:\Users\Tracy E. Miller\Downloads\OpenRocket\OpenRocket.New\OpenRocket-701-restore-optimizer.jar
  java.class.version=55.0
  java.home=C:\Program Files\Java\jdk-11.0.8
  java.io.tmpdir=C:\Users\TRACYE~1.MIL\AppData\Local\Temp\
  java.library.path=C:\Program Files\Java\jdk-11.0.8\bin;C:\WINDOWS\Sun\Java\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files\Java\jre1.8.0_251\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Users\Tracy E. Miller\AppData\Local\Microsoft\WindowsApps;;.
  java.runtime.name=Java(TM) SE Runtime Environment
  java.runtime.version=11.0.8+10-LTS
  java.specification.name=Java Platform API Specification
  java.specification.vendor=Oracle Corporation
  java.specification.version=11
  java.vendor=Oracle Corporation
  java.vendor.url=https://openjdk.java.net/
  java.vendor.url.bug=https://bugreport.java.com/bugreport/
  java.vendor.version=18.9
  java.version=11.0.8
  java.version.date=2020-07-14
  java.vm.compressedOopsMode=32-bit
  java.vm.info=mixed mode
  java.vm.name=Java HotSpot(TM) 64-Bit Server VM
  java.vm.specification.name=Java Virtual Machine Specification
  java.vm.specification.vendor=Oracle Corporation
  java.vm.specification.version=11
  java.vm.vendor=Oracle Corporation
  java.vm.version=11.0.8+10-LTS
  jdk.debug=release
  jnlp.jogamp.tmp.cache.root=jln7806350859711808468
  line.separator=\u000d\u000a
  os.arch=amd64
  os.name=Windows 10
  os.version=10.0
  path.separator=;
  sun.arch.data.model=64
  sun.awt.enableExtraMouseButtons=true
  sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler
  sun.boot.library.path=C:\Program Files\Java\jdk-11.0.8\bin
  sun.cpu.endian=little
  sun.cpu.isalist=amd64
  sun.desktop=windows
  sun.io.unicode.encoding=UnicodeLittle
  sun.java.command=C:\Users\Tracy E. Miller\Downloads\OpenRocket\OpenRocket.New\OpenRocket-701-restore-optimizer.jar
  sun.java.launcher=SUN_STANDARD
[... 311 lines cut ...]
1208     38.863 DEBUG  (OpenRocketSaver.java:272) Saving component Body tube
1209     38.863 DEBUG  (OpenRocketSaver.java:272) Saving component Inner Tube
1210     38.863 DEBUG  (OpenRocketSaver.java:272) Saving component Inner Tube
1211     38.863 DEBUG  (OpenRocketSaver.java:272) Saving component Transition
1212     38.863 DEBUG  (OpenRocketSaver.java:272) Saving component Body tube
1213     38.863 DEBUG  (OpenRocketSaver.java:272) Saving component Centering ring
1214     38.863 DEBUG  (OpenRocketSaver.java:272) Saving component Centering ring
1215     38.863 DEBUG  (OpenRocketSaver.java:109) Writing complete, flushing buffers
1216     38.917 DEBUG  (SwingWorkerDialog.java:165) Estimated run time, estimate=201 remaining=0
1217     39.017 INFO   (SwingWorkerDialog.java:150) Worker completed before opening dialog
1218     54.702 VBOSE  (RocketFigure3d.java:426) GL - init()
1219     54.704 VBOSE  (RocketFigure3d.java:456) GL - reshape()
1220     54.711 DEBUG  (TextureCache.java:119) Loading texture Texture [offset=(0.00000,0.00000,0.00000), center=(0.00000,0.00000,0.00000), scale=(1.00000,1.00000,0.00000), rotation=0.0, image=/datafiles/textures/spiral-wound-alpha.png]
1221     54.787 DEBUG  (TextureCache.java:119) Loading texture Texture [offset=(0.00000,0.00000,0.00000), center=(0.00000,0.00000,0.00000), scale=(1.00000,1.00000,0.00000), rotation=0.0, image=/datafiles/textures/wood.jpg]
1222     54.849 DEBUG  (TextureCache.java:119) Loading texture Texture [offset=(0.00000,0.00000,0.00000), center=(0.00000,0.00000,0.00000), scale=(1.00000,1.00000,0.00000), rotation=0.0, image=/datafiles/textures/wadding.png]
1223     54.871 DEBUG  (TextureCache.java:119) Loading texture Texture [offset=(0.13000,0.00000,0.00000), center=(0.00000,0.00000,0.00000), scale=(2.00000,0.38168,0.00000), rotation=0.0, image=decals/Markings.Virus.Rev.24.05.300-DPI.Sign.Grn.jpg]
1224     55.320 DEBUG  (TextureCache.java:119) Loading texture Texture [offset=(0.00000,-0.00000,0.00000), center=(0.00000,0.00000,0.00000), scale=(1.00000,1.00000,0.00000), rotation=0.0, image=decals/Plunger.02.png]
1225     55.436 DEBUG  (TextureCache.java:119) Loading texture Texture [offset=(0.15000,0.00000,0.00000), center=(0.00000,0.00000,0.00000), scale=(2.00000,1.04167,0.00000), rotation=0.0, image=decals/Elipse.02.03.Gray.png]
1226     55.474 DEBUG  (TextureCache.java:119) Loading texture Texture [offset=(0.00000,0.00000,0.00000), center=(0.00000,0.00000,0.00000), scale=(1.00000,1.00000,0.00000), rotation=0.0, image=/datafiles/textures/chute.jpg]
1227     55.505 DEBUG  (TextureCache.java:119) Loading texture Texture [offset=(0.00000,0.00000,0.00000), center=(0.00000,0.00000,0.00000), scale=(1.00000,1.00000,0.00000), rotation=0.0, image=/datafiles/textures/balsa.jpg]
1228     55.590 DEBUG  (RocketFigure3d.java:390) Redrawing Overlay
1229     76.996 USER   (BasicFrame.java:773) Bug report selected
1230     77.001 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------
[Anti-Virus.New OR.zip](https://github.com/openrocket/openrocket/files/4983347/Anti-Virus.New.OR.zip)

hcraigmiller, 2020-07-27:
Screenshots

[image]
[image]

hcraigmiller, 2020-07-27:
Photo Studio screenshot.

[image]

teyrana, 2020-08-08:
hey, @hcraigmiller 
I've seen this very rarely on my second monitor, but I'm having a hard time tracking down the root cause.... what sort of computer / hardware setup do you have? 

- single monitor? 
- laptop vs desktop?
- which video card manufacturer (I'm mostly interested in the drivers, here)

hcraigmiller, 2020-08-08:
Dell notebook, no external monitor. I have attached the system and display specifications. Is this the information you are looking for?

[image]
[image]

hcraigmiller, 2021-06-14:
Issue no longer exists.

closed 2021-06-14 by hcraigmiller