openrocket/openrocket #723: Bug Report -- Crash During Scaling

hcraigmiller, 2020-07-29:
Previously posted Anti-Virus .ork

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
net.sf.openrocket.util.BugException: BUG:   attempted to initialize an InertiaMatrix with a negative inertia value.
	at net.sf.openrocket.masscalc.RigidBody.<init>(RigidBody.java:24)
	at net.sf.openrocket.masscalc.MassCalculation.calculateStructure(MassCalculation.java:362)
	at net.sf.openrocket.masscalc.MassCalculation.calculateStructure(MassCalculation.java:316)
	at net.sf.openrocket.masscalc.MassCalculation.calculateStructure(MassCalculation.java:316)
	at net.sf.openrocket.masscalc.MassCalculation.calculateStructure(MassCalculation.java:316)
	at net.sf.openrocket.masscalc.MassCalculation.calculateAssembly(MassCalculation.java:267)
	at net.sf.openrocket.masscalc.MassCalculator.calculate(MassCalculator.java:114)
	at net.sf.openrocket.masscalc.MassCalculator.calculateLaunch(MassCalculator.java:82)
	at net.sf.openrocket.gui.scalefigure.RocketPanel.updateExtras(RocketPanel.java:605)
	at net.sf.openrocket.gui.scalefigure.RocketPanel.access$200(RocketPanel.java:87)
	at net.sf.openrocket.gui.scalefigure.RocketPanel$6.stateChanged(RocketPanel.java:281)
	at net.sf.openrocket.util.ComponentChangeAdapter.componentChanged(ComponentChangeAdapter.java:20)
	at net.sf.openrocket.rocketcomponent.Rocket.notifyAllListeners(Rocket.java:514)
	at net.sf.openrocket.rocketcomponent.Rocket.fireComponentChangeEvent(Rocket.java:474)
	at net.sf.openrocket.rocketcomponent.RocketComponent.fireComponentChangeEvent(RocketComponent.java:1761)
	at net.sf.openrocket.rocketcomponent.RocketComponent.fireComponentChangeEvent(RocketComponent.java:1773)
	at net.sf.openrocket.rocketcomponent.ComponentAssembly.setAxialOffset(ComponentAssembly.java:140)
	at jdk.internal.reflect.GeneratedMethodAccessor90.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at net.sf.openrocket.util.Reflection$Method.invoke(Reflection.java:34)
	at net.sf.openrocket.gui.dialogs.ScaleDialog$GeneralScaler.scale(ScaleDialog.java:561)
	at net.sf.openrocket.gui.dialogs.ScaleDialog.scale(ScaleDialog.java:489)
	at net.sf.openrocket.gui.dialogs.ScaleDialog.doScale(ScaleDialog.java:445)
	at net.sf.openrocket.gui.dialogs.ScaleDialog.access$600(ScaleDialog.java:48)
	at net.sf.openrocket.gui.dialogs.ScaleDialog$4.actionPerformed(ScaleDialog.java:395)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6400)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117)
	at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
[... 580 lines cut ...]
5605    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
5606    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
5607    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
5608    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
5609    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.AccessController.doPrivileged(Native Method)
5610    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
5611    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
5612    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
5613    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
5614    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.AccessController.doPrivileged(Native Method)
5615    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
5616    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
5617    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
5618    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
5619    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
5620    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
5621    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
5622    305.531 STDERR (SwingExceptionHandler.java:49) 	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
5623    305.531 INFO   (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
5624    305.531 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
5625    308.015 USER   (SwingExceptionHandler.java:220) User requested sending bug report
5626    308.027 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

[image]

teyrana, 2020-07-30:
We fixed a similiar issue with #330 / #395; 

Are you able to provide the .ork file that's causing this?   

(although this looks like a distinct bug -- it starts with: "attempted to initialize an InertiaMatrix with a negative inertia". )

hcraigmiller, 2020-07-30:
Here is the .ork file. Scaling error occurred when Airframe Payload Section was selected (highlighted) [4.024"] and the entire rocket was scaled to 2.6".
[Anti-Virus.New OR.zip](https://github.com/openrocket/openrocket/files/5004116/Anti-Virus.New.OR.zip)

teyrana, 2020-08-01:
## Reproducing

Using the same rocket ("Anti-Virus 4x"), and the following dialog inputs... 
[image]

I see a correctly scaled rocket with no crash, and no exceptions in the logs.

It _does_ show excessive scaling calls, but that doesn't seem to be the issue referenced above.

teyrana, 2020-08-01:
## Reproduce

On the second try, using the exact scale numbers generates the exception:
[image]

So, this is not a general bug, it only happens at certain value(s?) of scaling.

teyrana, 2020-08-01:
So, this is not a general bug, it only happens at certain value(s?) of scaling.

1. Okay %
    - 150
    - 90
2. Bad %
    - 47.27272727272728

hcraigmiller, 2020-08-01:
Using integers, anything less than 70% triggers an error. You can use two
scales to go from 4.024” to 2.6” without an error.

hcraigmiller, 2020-08-01:
This is not the only design that will not scale from 4.024” to 2.6” without
throwing an error, so it does not appear to be design specific.

teyrana, 2020-08-01:
> This is not the only design that will not scale from 4.024” to 2.6” without throwing an error, so it does not appear to be design specific.

Fair enough.  I'm not saying it is.  The whole process of reproducing the error, is not about fixing the rocket design, it's about quickly iterating through a bunch of debugging steps to find the root cause. :)

hcraigmiller, 2020-08-01:
Would it be helpful to have a second .design ork file that also errors when scaling?

teyrana, 2020-08-01:
> Would it be helpful to have a second .design ork file that also errors when scaling?

In this case, that's probably not necessary. 

The fix is incoming, so try _that_ branch on however many rockets you like ;)

commit 908cd3e6 (2020-08-02): Merge pull request #728 from teyrana/fix/723-scaling

[Fixes #723] Fix scaling crashes in some circumstances
closed 2020-08-02 by teyrana