openrocket/openrocket #727: UI: Object configuration dialogs have gotten larger

neilweinstock, 2020-07-30:
I just noticed that the object configuration dialogs have gotten somewhat larger in the new builds due to expanded spacing around the header and footer:
[image]

This is not a major issue but to me it is better the old way.  The extra spacing does not (IMHO) enhance usability, it just makes the dialogs take up more space on the screen.

teyrana, 2020-08-01:
Hmmm, I don't remember anyone explicitly setting this.

Which dialogs are you seeing it on?

neilweinstock, 2020-08-01:
I think it's all of the component configuration dialogs.

teyrana, 2020-08-01:
So, to be clear, you haven't seen it in dialogs _besides_ the Component Configuration ones?

teyrana, 2020-08-01:
looks like it was a side effect of:
 https://github.com/openrocket/openrocket/issues/424

commit ed409204 (2020-08-01): Merge pull request #730 from teyrana/fix/727/config-spacing

[fixes #727][gui][cosmetic] Adjusted layout in RocketComponentConfig layout
closed 2020-08-01 by teyrana