openrocket/openrocket #741: Rail Buttons Do Not Display on Pod Body Tubes

hcraigmiller, 2020-08-09:
Rail buttons that display on the main body tube disappear when moved to a pod set body tube. A change in the CG shows the move as expected, but the CP does not change. 

[Rail Button.Pod Set.No Display.zip](https://github.com/openrocket/openrocket/files/5048488/Rail.Button.Pod.Set.No.Display.zip)

hcraigmiller, 2020-08-09:
Correction, the rail buttons were just out of view; the display did not reduce to show the rail buttons that were out of visual range.

teyrana, 2020-08-09:
1.  Like launch lugs, rail buttons are not aerodynamic, and will not affect CP.

2. What do you mean by "display reduce”?   Auto zoom?

hcraigmiller, 2020-08-09:
It did not auto zoom out to show the rail buttons.

teyrana, 2020-08-09:
Dude... how big are your rail buttons ?!
(Mostly rhetorical)

hcraigmiller, 2020-08-09:
Just looking for program functionality, not actual designing a rocket. I did find an irregularity in function, when comparing what happens with launch lugs and rail buttons, when moved from the main body tube to a pod body tube, a launch lug stays on the body tube, when you do the same with rail buttons, the buttons are some distance from the pod body tube.

[image]
[image]

teyrana, 2020-08-16:
## Breadcrumbs
1. Reproduce
2. Observed on side views AND rear views

commit 8a14d98f (2020-08-16): [fixes #741] RailButtons will correctly identify their first parent BodyTube
closed 2020-08-16 by teyrana