openrocket/openrocket #746: Crazy fin CP calculation (seems like a numerical error)

neilweinstock, 2020-08-15:
Found this in 15.03, it persists in the current unstable.  This one took me quite a while to isolate.  It bit me badly on a rocket I'm building right now.

In the following set of four screenshots, I've edited the Y value of the second to last point on the fin by .001" each time (value shown in the magnifier).  Look at how the CP bounces around crazy from those tiny insignificant changes.  In addition to being way larger CP variations than would be justified, the movement isn't even monotonic.  This suggests to me a numerical error of some sort.

The trigger for the problem might be the spikes on the fins.  In any case, the numbers should not behave this way.

Here's the screenshot(s):
[image]

Here's the design:
[crazy_cp.ork.zip](https://github.com/openrocket/openrocket/files/5078013/crazy_cp.ork.zip)

JoePfeiffer, 2020-08-16:
Ouch!  Yes, that's got to be a bug....  after a brief look, seems like something is going on with the midchord angle.

neilweinstock, 2020-08-16:
That's a good one, isn't it? :)

I've played around with the spikes a fair bit, and can't establish a clear connection.  But without the spikes things seem to calm down, so they're involved somehow.

JoePfeiffer, 2020-08-17:
It was the presence of a nearly-horizontal line in the fin.  When the fin was split up into "slices" to calculate its geometry, the calculation of the intersection between the line and the bounds of a slice were far, far outside the bounds of the fin itself.

neilweinstock, 2020-08-18:
Interesting!  Got a fix?

wcravens, 2020-08-18:
https://www.amazon.com/Chaos-Making-Science-James-Gleick/dp/0143113453 :-D

Wes

PR #750 (2020-08-19): clamp x coordinate of intersections
In FinSetCalc.calculateFinGeometry(), make sure x coordinates of intersections between lines defining fins and strips being used to estimate geometry aren't beyond bounds of fin.
Fixes #746
closed 2020-08-19 by JoePfeiffer