openrocket/openrocket #747: Crash on startup - mac osx Catalina

SamyTahar, 2020-08-15:
Hi Team,

The application is crashing just after the splash screen. 

Do you know what could be the issue? 

I have tried to build the app from the source code and it did build successfully, however, when I am launching the app from the folder core/build/jar/OpenRocket-Core.jar I have an error that says :

[image]

Thanks for your help,
Samy

JoePfeiffer, 2020-08-16:
In your first remark (The application is crashing just after the splash screen), are you using the distributed jar file for version 15.03?  If so, the most likely problem is that your computer is using a version of Java more recent than Java 8.  Unfortunately, 15.03 can not run with more recent versions.  The easiest way forward in this case (for mac) is the prepackaged installer at https://www.dropbox.com/s/33o53vlle5b0uex/OpenRocket-15.03.dmg?dl=0

When you built from source code, was that from the code at the release page?  If so, you'll come up against the same problem with Java versions.  Current development is on the unstable branch, which does build and run under modern Java.  However, in that case after building you should find a jar file at .../swing/build/jare/OpenRocket.jar, which is what you'll need to run.

SamyTahar, 2020-08-16:
Thanks Joe,

The prepackage app is working well, even if the java version is 14 installed. I tried to launch the app in the swing/build/jar folder with the jar launcher short cut the app is crashing however when I am doing it in command line `java -jar OpenRocket.jar` it is working well ( the the unstable branch.

madsmtm, 2020-08-18:
@JoePfeiffer would you mind documenting how you obtained the prepackaged installer? It's a bit unnerving to get something like this from an unknown source.

And also, if it's indeed official, why don't we advertise this in the README?

JoePfeiffer, 2020-08-18:
> @JoePfeiffer would you mind documenting how you obtained the prepackaged installer? It's a bit unnerving to get something like this from an unknown source.
> 
> And also, if it's indeed official, why don't we advertise this in the README?

The Windows and OSX installers were built by Neil Weinstock, another of the devs.  Hmmm, not sure why his OSX installer wasn't on the download page; I just added it there and modified the description of the release a bit to call out the need for Java 8. I built the Linux appimage; a nice thing about appimages is you can take them apart to verify the files in them are unaltered.

I'm really reticent to make any changes to the release itself after the fact.

closed 2020-10-27 by JoePfeiffer