openrocket/openrocket #748: Uncaught Exception Error Window Will Not Close Without Ending Task (Windows 10)

hcraigmiller, 2020-08-16:
In the "Uncaught Exception" error window, you can select "View bug report" to obtain the report, and you can close the bug report window, but the "Uncaught Exception" window will not close with either the "Close" button or the "X" button. When either is used,  the "Uncaught Exception" window momentarily disappears but almost immediately reappears. The only way to get out of this loop is to use the Task Manager and end the Java Platform task. The Bug Report is attached as a .txt file.

[Bug Report.txt](https://github.com/openrocket/openrocket/files/5080090/Bug.Report.txt)

The error can be reproduced by opening the attached .ork file (Neil's base file), then moving the marked transition from Stage 1 to Stage 2.

[plasma_dart_2stage.Rev.01.Error.zip](https://github.com/openrocket/openrocket/files/5080102/plasma_dart_2stage.Rev.01.Error.zip)
[image]

If more than one instance of OR is open, this error prevents you from selecting any other instance.

teyrana, 2020-08-16:
@hcraigmiller 

Just FYI, the behavior surrounding this dialog is not consistent. Nor even predictable -- the behavior of that window depends on the underlying trigger exception. Sometimes you can continue working fine, and other times it's a disaster.

The official answer is that as a bug, we do not guarantee anything after this diaog appears.
Anytime it does appear, we consider that a bug.  If it's a recoverable exception, then that recovery has already been programmed in, and you won't see this window. 

If you see this window, please:
1. report the bug (as you've done here)
2. Restart the entire program.

teyrana, 2020-08-16:
@hcraigmiller 
I couldn't reproduce this crash -- could you please confirm which build you're running this one? 
Specfically, that it's after PR #744, which fixes a related bug?

hcraigmiller, 2020-08-16:
Using pre-built jar #701, I guess I'm way out of date...

hcraigmiller, 2020-08-16:
If its not too much trouble, could I get a jar of the current unstable?

teyrana, 2020-08-27:
closed as resolved.

closed 2020-08-27 by teyrana