openrocket/openrocket #751: Inconsistent terminology used for angular/radial component positioning

hcraigmiller, 2020-08-19:
Launch lugs use "Radial position" and rail buttons use "Angular position." Mass components and inner tubes use radial position, but pods use "Angle." Should the positioning be standardized to either angular or radial positioning?

neilweinstock, 2020-08-20:
Yes, should be standardized.  I'm not sure what the preferred term should be.

Also, fin sets simply use "Angle", although *maybe* it's a different enough thing to justify a different term, not sure.

Inner tubes have their own separate tab for "Radial Position", and inside that tab they have "Radial Direction" which is yet another term for the same thing.  It's debatable whether that should be a separate tab at all, but either way consistency in terminology would be good.

teyrana, 2020-11-02:
Since I'm not seeing any replies, I'm unifying the labels to "Rotation".  This matches LaunchLugs and RailButtons to the Fin Config Dialogs.

neilweinstock, 2020-11-02:
That seems good to me. 👍

commit 39c29e8c (2020-11-04): Merge pull request #805 from teyrana/fix/801/launch-lug-radial-position

[fixes #801][fixes #751][fixes #752] Fixes multiple issues with LaunchLug component & ConfigDialog
commit cf8b4022 (2020-11-04): [fixes #751] Unifies angle-offset labels to 'Rotation' for component config dialogs
closed 2020-11-04 by teyrana