openrocket/openrocket #759: openrocket crashes on opening on my MacBook pro

groupracer, 2020-08-26:
Process:               java [1763]
Path:                  /Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java
Identifier:            net.java.openjdk.cmd
Version:               1.0 (1.0)
Code Type:             X86-64 (Native)
Parent Process:        Jar Launcher [1762]
Responsible:           java [1763]
User ID:               501

Date/Time:             2020-08-25 22:21:17.857 -0400
OS Version:            Mac OS X 10.14.6 (18G6020)
Report Version:        12
Bridge OS Version:     3.0 (14Y906)
Anonymous UUID:        CEF809BD-796A-C6D4-9EB7-95EF9BFBA207

Sleep/Wake UUID:       53277ACA-F393-45BD-ACD7-4F771D918910

Time Awake Since Boot: 10000 seconds
Time Since Wake:       1300 seconds

System Integrity Protection: enabled

Crashed Thread:        33  Java: AWT-EventQueue-0

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x8:
--> 
    __TEXT                 000000010d577000-000000010d586000 [   60K] r-x/rwx SM=COW  /Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
terminating with uncaught exception of type NSException
abort() called

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff514139ad __exceptionPreprocess + 256
1   libobjc.A.dylib                     0x00007fff7bb31a17 objc_exception_throw + 48
2   CoreFoundation                      0x00007fff5142d50d -[NSException raise] + 9
3   AppKit                              0x00007fff4e9d25ca -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 317
4   AppKit                              0x00007fff4e9cf9f7 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1479
5   AppKit                              0x00007fff4e9cf42a -[NSWindow initWithContentRect:styleMask:backing:defer:] + 45
6   libnativewindow_macosx.jnilib       0x00000001172986be Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 382
7   ???                                 0x0000000117b8f6c7 0x0 + 4692965063

Thread 0:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff7d43521a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff7d435768 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff5135d882 __CFRunLoopServiceMachPort + 328
3   com.apple.CoreFoundation      	0x00007fff5135cdf0 __CFRunLoopRun + 1612
4   com.apple.CoreFoundation      	0x00007fff5135c552 CFRunLoopRunSpecific + 455
5   com.apple.HIToolbox           	0x00007fff505bb1ab RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff505baee5 ReceiveNextEventCommon + 603
7   com.apple.HIToolbox           	0x00007fff505bac76 _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff4e95277d _DPSNextEvent + 1135
9   com.apple.AppKit              	0x00007fff4e95146b -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1361
10  libosxapp.dylib               	0x00000001135b9ab8 -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 121
11  com.apple.AppKit              	0x00007fff4e94b588 -[NSApplication run] + 699
12  libosxapp.dylib               	0x00000001135b98fd +[NSApplicationAWT runAWTLoopWithApp:] + 222
13  libsplashscreen.dylib         	0x000000011359ab86 __SplashInitPlatform_block_invoke + 70
14  JavaNativeFoundation          	0x00007fff55496fce +[JNFRunLoop _performCopiedBlock:] + 17
15  com.apple.Foundation          	0x00007fff536572b2 __NSThreadPerformPerform + 328
16  com.apple.CoreFoundation      	0x00007fff51379d13 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
17  com.apple.CoreFoundation      	0x00007fff51379cb9 __CFRunLoopDoSource0 + 108
18  com.apple.CoreFoundation      	0x00007fff5135d67f __CFRunLoopDoSources0 + 195
19  com.apple.CoreFoundation      	0x00007fff5135cc49 __CFRunLoopRun + 1189
20  com.apple.CoreFoundation      	0x00007fff5135c552 CFRunLoopRunSpecific + 455
21  java                          	0x000000010d57c312 CreateExecutionEnvironment + 894
[... 932 lines cut ...]
__DATA                            66.3M      292 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       230.7M       24 
__TEXT                           207.7M      285 
__UNICODE                          564K        1 
mapped file                        1.2G      370 
shared memory                      672K       11 
===========                     =======  ======= 
TOTAL                              7.5G     1288 
TOTAL, minus reserved VM space     7.4G     1288 

Model: MacBookPro14,3, BootROM 428.0.0.0.0, 4 processors, Intel Core i7, 3.1 GHz, 16 GB, SMC 2.45f0
Graphics: kHW_IntelHDGraphics630Item, Intel HD Graphics 630, spdisplays_builtin
Graphics: kHW_AMDRadeonPro560Item, Radeon Pro 560, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x173), Broadcom BCM43xx 1.0 (7.77.61.3 AirPortDriverBrcmNIC-1305.10)
Bluetooth: Version 6.0.14d11, 3 services, 27 devices, 1 incoming serial ports
Network Service: AirPort, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: iBridge
USB Device: iPhone
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.1
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.1

wolsen, 2020-08-26:
What version of OpenRocket are you running? How are you running OpenRocket? What's the java version that is in use?

Java version can be determined by running `/Library/Internet` Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java -version` from a command line.

As a note, the released version of Openrocket (OpenRocket 15.03) is only compatible with java 8. Unfortunately, with the information provided I cannot find the OpenRocket version nor the Java version in use

groupracer, 2020-08-26:
Hi,
thanks for the response. Sorry about not including version etc, but i hit the send button too soon. i figured when i get a response i’ll send that info. so…

I am running 15.03, as  downloaded on a MacBook pro. I’m still using Mojave. Java is version 8 update 261 on 8/26/20. I tried running your script but it does not exist on my mac.

I was able to run it ok but it started to crash when i changed to solid views within the program. I was able to open and run it several weeks ago but it did exhibit the same crashes.

I’ve been running OpenRocket from the app in applications.

Is this what you need? please let me know if there is anything else i may need to send you.

Regards,

Evan

neilweinstock, 2020-08-31:
Go to Edit->Preferences and check your graphics settings.  They should look like this:
[image]

groupracer, 2020-08-31:
thanks but i cannot even open the app it keeps crashing when i launch it.

neilweinstock, 2020-08-31:
Ah.  Have you tried the packaged installer?  If not: https://www.rocketryforum.com/threads/openrocket-packaged-installers-for-windows-and-mac-to-solve-all-your-java-problems.143540/

No guarantees but it solves a lot of problems.

groupracer, 2020-08-31:
hi,
tried it  and now it works as before. i don’t know what happened, but at least i can work with my files.  it still crashes when i use the rendering mode, but it seems that the files that crash it are ones that i made. oh well at least i can use it now.

Thanks!

neilweinstock, 2020-09-01:
So, can we close this one?  I don't have permission.

groupracer, 2020-09-01:
Yes we can. Thanks for the help. I will have to close it from my end?

neilweinstock, 2020-09-01:
Yup, just hit "close" or "comment and close", as appropriate.

groupracer, 2020-09-01:
thanks for the help!

closed 2020-09-01 by groupracer

broofa, 2021-05-24:
Please consider reopening.

I just went through the "obvious" path for installing OpenRocket...
 * Google "openrocket" (takes me to `openrocket.info` site)
 * Click the big fat "Download 15.03" button
 * Click  the downloaded `OpenRocket-15.03.jar` file to open it
 * _Splash panel appears briefly then..._ **crash**

Crash reporter pops up with [this crash report](https://gist.github.com/broofa/6ead968aeae8d5cd5a415a6cde656f58).

At a minimum, users going through this flow should be warned that this issue exists and given a pointer to the [Github releases page](https://github.com/openrocket/openrocket/releases) where they can get the `.dmg` download (that fixes this issue, as suggested by @neilweinstock ).

## system info

FWIW, this is on a new(ish) 13" Macbook PRO w/ M1 processor.

The "About" screen for Java (as shown in System Preferences): 
[image]

FWIW, here's what `npx envinfo`  (node.js-base utility for dumping system info)has to say about my system:
```
  System:
    OS: macOS 11.1
    CPU: (8) x64 Apple M1
    Memory: 155.18 MB / 16.00 GB
    Shell: 5.1.0 - /usr/local/bin/bash
  Binaries:
    Node: 14.16.1 - ~/.nvm/versions/node/v14.16.1/bin/node
    npm: 6.14.12 - ~/.nvm/versions/node/v14.16.1/bin/npm
  Managers:
    Homebrew: 3.1.5 - /usr/local/bin/brew
    pip2: 9.0.1 - /usr/local/bin/pip2
    RubyGems: 3.0.3 - /usr/bin/gem
  Utilities:
    CMake: 3.12.2 - /usr/local/bin/cmake
    Make: 3.81 - /usr/bin/make
    GCC: 4.2.1 - /usr/bin/gcc
    Git: 2.29.2 - /usr/local/bin/git
    Clang: 12.0.5 - /usr/bin/clang
    FFmpeg: 4.3.2 - /usr/local/bin/ffmpeg
  Servers:
    Apache: 2.4.46 - /usr/sbin/apachectl
  Virtualization:
    Docker: 20.10.0 - /usr/local/bin/docker
  SDKs:
    iOS SDK:
      Platforms: iOS 14.5, DriverKit 20.4, macOS 11.3, tvOS 14.5, watchOS 7.4
  IDEs:
    Nano: 2.0.6 - /usr/bin/nano
    Vim: 8.2 - /usr/bin/vim
    Xcode: 12.5/12E262 - /usr/bin/xcodebuild
  Languages:
    Bash: 5.1.0 - /usr/local/bin/bash
    Go: 1.11.4 - /usr/local/bin/go
    Java: 1.8.0_144 - /usr/bin/javac
    Perl: 5.28.2 - /usr/bin/perl
    PHP: 7.3.24 - /usr/bin/php
    Python: 2.7.16 - /usr/bin/python
    Python3: 3.9.1 - /usr/local/bin/python3
    Ruby: 2.6.3 - /usr/bin/ruby
  Databases:
    MySQL: 5.7.32 - /usr/local/opt/mysql@5.7/bin/mysql
    SQLite: 3.32.3 - /usr/bin/sqlite3
  Browsers:
    Chrome: 90.0.4430.212
    Firefox: 88.0.1
    Safari: 14.0.2
```

neilweinstock, 2021-05-24:
>     MySQL: 5.7.32 - ***@***.***/bin/mysql

Use the packaged installer:
https://github.com/openrocket/openrocket/releases/download/release-15.03/OpenRocket-15.03.dmg

mgd722, 2021-08-06:
Seconded. The obvious path to getting this running on my system is to click the big "download now" button, but that failed without saying why. If I wasn't a software dev and knew what to look for, I'd be dead in the water on this one.