openrocket/openrocket #761: Motor configuration changes when saved and reopened (specific to Aerotech G80)

hcraigmiller, 2020-08-26:
This may be specific to the Aerotech 29mm G80 motor. When a configuration containing this motor is saved, it changes to a G78 when reopened. I tried a number of different configurations, but this is the only motor that kept changing.

[image]

On a side note, new configurations are not always created in the same order (such as the beginning or the end of the existing list).  How are new configurations ordered?

teyrana, 2020-10-18:
@hcraigmiller 
Please post the file for debugging

Even the "wrong" one, as it will save a lot of time to reproduce the issue

hcraigmiller, 2020-10-19:
Design file is attached:
[LOC.T-Loc.Error.Motor.G80.zip](https://github.com/openrocket/openrocket/files/5405243/LOC.T-Loc.Error.Motor.G80.zip)

Open the design file, and select Motors and Configuration.
[image]

JoePfeiffer, 2020-11-09:
Confirmed.  Hmmm....

JoePfeiffer, 2020-11-10:
Hash collision!  Turns out both motors have the same "digest" (effectively a hash), and there's no check for a collision.  Working on it...

closed 2020-12-01 by JoePfeiffer