openrocket/openrocket #763: Selection of deployment configuration for one motor on recovery tab opens the config menu for a different motor

hcraigmiller, 2020-08-27:
Open the attached OR design file.
[MDRM.Fighter.Rev.02.zip](https://github.com/openrocket/openrocket/files/5133223/MDRM.Fighter.Rev.02.zip)

Go to the Motors & Configurations tab and click on the Recovery tab. On the Recovery tab, select the Main Parachute, Apogee setting for the H115 motor (either double-click on the word "Apogee" or single-click on the word "Apogee" and click the "Select deployment" button). The Select Deployment Configuration window shows that the recovery configuration for the H115 motor is being modified. Click OK or cancel to exit.

Now, select the Main Parachute, Apogee setting for the G138 motor. The Select Deployment Configuration window shows that _the recovery configuration for the **H115** motor is still being modified_.  Change deployment to first ejection charge of the stage then click OK, and it confirms that the recovery configuration for the H115 motor, not the G138 motor, was modified.

[image]

Even if you go straight from opening the design file to selecting the Main Parachute, Apogee setting for the G138 motor. The Select Deployment Configuration window will show that the recovery configuration for the **H115** motor is being modified.

JoePfeiffer, 2020-12-12:
The underlying issue is the selected configuration isn't being kept in sync between the Motors tab, Recovery tab, Stages tab, and the main rocket display window. When selecting a motor it uses the configuration selected in the pane so it isn't as noticeable; when selecting recovery options it's really noticeable since the configuration selected in the main display window determines which configuration is changed, which is not expected.

15.03 had this right...

hcraigmiller, 2020-12-13:
May be related to #771 and/or #845.

JoePfeiffer, 2020-12-13:
I *think* they're separate issues.  But I'll check.

SiboVG, 2021-06-13:
From the first looks, it seems that the pop-up window uses the motor that is selected in the flight configuration. Selecting G138-5 in flight configuration makes the G138-5 to be edited in the deployment configuration, in both cases, both when selecting the G138-5 and H115-14 in the Recovery window. Will look into fixing it.

hcraigmiller, 2021-06-13:
Your observation appears correct. In fact, if you select "Flight configuration: No Motors" then you cannot "override" the selected recovery configuration for any selection (the change is rejected); you can only change a configuration by using the "Change all configuration using the default deployment event" selection, changing all recovery configurations, not just the one you selected.

SiboVG, 2021-06-14:
I might have found a possible solution for it by passing the FlightConfigurationId to the DeploymentSelectionDialog constructor. The ID is passed in selectDeployment from RecoveryConfigurationPanel, which in turn is extracted from the currently selected item in the recovery table. The current implementation took the FlightConfigurationId from the rocket flight configuration, but I changed that to be the ID of the configuration selected in the recovery table. Will do a pull request as soon as possible, but after editing these changes, it worked perfectly on my version.

hcraigmiller, 2021-06-14:
Uploading a test .jar file may be helpful for testing and verification purposes.

hcraigmiller, 2021-06-14:
This issue was fixed by PR #943.

closed 2021-06-14 by hcraigmiller

JoePfeiffer, 2021-06-14:
#942 was merged, not #943.  Did that also fix this issue?

hcraigmiller, 2021-06-15:
Yes it did.