openrocket/openrocket #768: Strange flight sim behavior with 3-stage design

neilweinstock, 2020-08-31:
There's a lot of weirdness here that is hard to explain.  I will try to do it with pictures.  There may be multiple issues here, but at this point I have no idea how to separate them.

Here is the design in question:
[Talos Terrier Recruit.ork.zip](https://github.com/openrocket/openrocket/files/5147969/Talos.Terrier.Recruit.ork.zip)

Here is 15.03 on left, latest unstable on right:
[image]
**_First, note that the apogee is *way* higher on 15.03._**

Next, note that 15.03 has a sim error: "Simulation values exceeded limits.  Try selecting a shorter time step".  This seems to be a common problem in multi-stage designs in 15.03 that is fixed in unstable.  Still, as I'll show below, the error seems to occur at apogee, so the apogee estimate looks reasonable.

**_Next, note that the unstable has a different sim error: "Recovery device opened while motor is still burning".  I *think* this is because the top stage sim is wacko (read on), but the plots do not show this happening._**  

Let's look at the plots (again, 15.03 on left, unstable on right).  I included lateral velocity just to ensure that everything was pretty stable and flying vertically:
[image]
The 15.03 plot looks pretty clean, other than that the plot stops at the top stage apogee, when the sim crashes apparently.

The unstable plot, on the other hand is... not so much.  The apogee of each stage is much lower than before.  Also, the top stage hits apogee almost immediately after ignition.

**_I cannot offer any reasonable hypothesis about what's going on here, other than "it's not working right"._**

Finally: note that the top stage is mostly base-drag-stabilized; either way, it is showing good stability margin and OR is definitely simulating it as if it is stable.  Nonetheless, I don't know if that is somehow contributing to the problem here.  But 15.03 has no problem with it.

Hope someone can sort out this mess! :)

JoePfeiffer, 2020-08-31:
The "recovery device opened" warning looks like #762, which is addressed in PR #765 (not yet merged).  Don't know about the other issues (yet).

neilweinstock, 2020-08-31:
Yup, it sure does.

JoePfeiffer, 2020-09-03:
It appears as if all three motors are igniting at launch (and I've verified that they aren't set to do that, and what's more ignition events are being inserted in the data export appropriately).  Still looking for why...

JoePfeiffer, 2020-09-07:
The motor thrust was calculated correctly, but mass was calculated as if they all ignited at t=0.

That was a problem, but not "the" problem.  It looks like the big problem was drag is being calculated as if all stages were still attached, throughout the flight.

neilweinstock, 2020-09-07:
Glad you have found the problems, those are pretty serious stuff.

JoePfeiffer, 2020-09-07:
Says something about how much the developers work with multistage rockets!

PR #789 (2020-10-10): Last fix on issue 768
On multistage rockets, there can be a next symmetric component that is not active.  Check for that.

I think this is the last fix required to close #768.
closed 2020-10-10 by teyrana