openrocket/openrocket #772: Component mass override of zero causes miscalculation of stage CG, even when stage mass override is set.

hcraigmiller, 2020-09-03:
Using the 09/02/2020 evening code update, a component mass override of zero causes a miscalculation of the stage CG, even when the mass override for that stage is set to a number greater than zero, and even though the component CG override is set to the component midpoint.

The attached design file uses three identical 8" tubes as stages and has all component and stage mass overrides set to 1, all component CG overrides set to 4", and the stage CG overrides set to 4", 12", and 20" (because the stage CG is being measured from the tip of Stage 1, not the top of the selected stage):
[Staged Tubes.Analysis.CG-4.MO-1.zip](https://github.com/openrocket/openrocket/files/5166373/Staged.Tubes.Analysis.CG-4.MO-1.zip)

Start by turning off all of the stage CG overrides, then turn off all of the stages. Turning on each stage individually results in a mid-stage CG of 4", 12", and 20". This is the same result you get if you also turn off all of the stage mass overrides, and the same result you get if you also turn off all of the component CG overrides. Everything matches and all is as expected.

However, with all stages off except Stage 1, if you set the component mass override of the Stage 1 body tube to zero, the Stage 1 CG computation results in “NA” even though the component CG override is set to 4"; the same is true for the other stages if the component masses are set to zero. Here, when a component mass override is set, the component CG override should be the stage CG result. Even if not set, theoretically, everything has a center of gravity, regardless of how infinitesimally small its mass may be. And, a component’s CG should be the same, regardless of how large or small its mass; the CG of a uniform 8" tube should be the same 4" CG whether it weighs an ounce or is so light that its mass is effectively zero.

If the computation is ignoring the component CG when overridden because the component mass override is set to zero, that omission may explain why the stage CG computation is incorrect even though the stage mass override is set to a value greater than zero.

I see two ways of handling this, (1) ensure that the stage mass override completely negates the component mass overrides within the stage, or (2) substitute the smallest value allowed in the place of a zero component mass override value, such as .001. The latter may be the easiest to implement by simply changing the mass override bounds so that the default values are changed from zero to .001, for example, and so that a mass override setting of zero is not allowed.

Just my perspective. What do others think?

teyrana, 2020-09-04:
@hcraigmiller 
Would you please clarify which commit you're using to test?  There is ongoing work on PR #769 for what sounds like the exact same issue.

hcraigmiller, 2020-09-04:
After merge of pull request #770 from teyrana/fix/758/mass-comp-render (0289d36) 22 hours ago.

closed 2020-09-05 by hcraigmiller