openrocket/openrocket #783: Issue: Windows 10, Java 1.8.0_261 and Java SDK 14.0.2 - OpenRocket encountered an uncaught exception

airqualityanthony, 2020-09-17:
Receiving the following when opening. 

OpenRocket encountered an uncaught exception. This typically signifies a bug in the software. 
    RuntimeException: Error starting OpenRocket

Please take a moment to report this bug to the developers. 
This can be done automatically if you have an internet connection. 

```
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.RuntimeException: Error starting OpenRocket
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:40)
	at net.sf.openrocket.startup.OpenRocket.main(OpenRocket.java:32)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at net.sf.openrocket.startup.jij.JarInJarStarter.runMain(JarInJarStarter.java:38)
	... 1 more
Caused by: java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1367)
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1342)
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1480)
	at net.sf.openrocket.startup.SwingStartup.main(SwingStartup.java:76)
	... 6 more
Caused by: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:553)
	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:419)
	at com.google.inject.internal.util.$CustomConcurrentHashMap$ComputingImpl.get(CustomConcurrentHashMap.java:2041)
	at com.google.inject.internal.FailableCache.get(FailableCache.java:50)
	at com.google.inject.internal.ConstructorInjectorStore.get(ConstructorInjectorStore.java:49)
	at com.google.inject.internal.ConstructorBindingImpl.initialize(ConstructorBindingImpl.java:125)
	at com.google.inject.internal.InjectorImpl.initializeJitBinding(InjectorImpl.java:521)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:847)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
	at com.google.inject.internal.FactoryProxy.notify(FactoryProxy.java:46)
	at com.google.inject.internal.ProcessedBindingData.runCreationListeners(ProcessedBindingData.java:50)
	at com.google.inject.internal.InternalInjectorCreator.initializeStatically(InternalInjectorCreator.java:133)
	at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:106)
	at com.google.inject.Guice.createInjector(Guice.java:95)
	at com.google.inject.Guice.createInjector(Guice.java:72)
	at com.google.inject.Guice.createInjector(Guice.java:62)
	at net.sf.openrocket.startup.SwingStartup.runInEDT(SwingStartup.java:143)
	at net.sf.openrocket.startup.SwingStartup.access$000(SwingStartup.java:43)
	at net.sf.openrocket.startup.SwingStartup$1.run(SwingStartup.java:79)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:306)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory
[... 246 lines cut ...]
  92      0.673 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:42)
  93      0.673 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.MembersInjectorStore$1.create(MembersInjectorStore.java:39)
  94      0.673 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.FailableCache$1.apply(FailableCache.java:39)
  95      0.673 STDERR (SwingExceptionHandler.java:49) 	at com.google.inject.internal.util.$MapMaker$StrategyImpl.compute(MapMaker.java:549)
  96      0.673 STDERR (SwingExceptionHandler.java:49) 	... 44 more
  97      0.673 STDERR (SwingExceptionHandler.java:49) Caused by: java.lang.ClassNotFoundException: javax.script.ScriptEngineFactory
  98      0.673 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:435)
  99      0.673 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:589)
 100      0.673 STDERR (SwingExceptionHandler.java:49) 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
 101      0.673 STDERR (SwingExceptionHandler.java:49) 	... 66 more
 102      0.673 INFO   (SwingExceptionHandler.java:64) Exception handler not on EDT, invoking dialog on EDT
 103      0.674 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
 104      6.409 USER   (SwingExceptionHandler.java:220) User requested sending bug report
 105      6.416 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------
```

I'm on Java SDK 14.0.2. 
JAVA 1.8.0_261

Windows 10. 

Thanks,

Anthony

wolsen, 2020-09-19:
This was responded to on the mailing list, but for completeness.

It appears its running as Java 14.0.2, which unfortunately doesn't work with OpenRocket 15.03.

The easiest way to get this to run is by using the [packaged windows installer](https://github.com/openrocket/openrocket/releases/download/release-15.03/OpenRocket-15.03-installer.exe).

An alternative is to ensure that your Java 8 is the runtime you are using. If you try to run OpenRocket.jar and get this crash, you can run java -version from a windows command line to see which version you are running. If it says something greater than Java 8/1.8 then you'll need to make sure that Java 8 is used when running the application.

wolsen, 2020-09-19:
I'm closing this out, but please do let us know if this doesn't work on Java 8.

closed 2020-09-19 by wolsen