openrocket/openrocket #788: Display smaller in current than 15.03

hcraigmiller, 2020-09-29:
The display windows are significantly smaller in the current unstable than in 15.03. The top image is a screenshot from 15.03 and the bottom is from the current unstable.

[image]

With regard to the Side view, it appears that the difference is attributable to the ruler bar, so the Side view display between the two versions is essentially the same.

But, the current unstable 3D image (the bottom image below) is much smaller than in 15.03 (the top image below), and the size reduction is not attributable to the window height. Moreover, in the current unstable the CG and CP markers are above the rocket, not on the rocket (the bottom image below).

[image]

Even lowering the screen resolution, the current unstable 3D Finished view is smaller than in 15.03.

[image]

wolsen, 2020-12-13:
The problem is that when setting up the view for the RocketFigure in 3d, the view size is doubling the span of the Y axes for the bounds required. This is likely legacy code, since the new bounds information takes this accurately into account and the doubling is no longer needed.

wolsen, 2020-12-13:
> The problem is that when setting up the view for the RocketFigure in 3d, the view size is doubling the span of the Y axes for the bounds required. This is likely legacy code, since the new bounds information takes this accurately into account and the doubling is no longer needed.

Note that this only applies to the scaling of the rocket figure displayed, not the scaling of the 3d viewport itself (which is #719)

PR #847 (2020-12-19): Fix rocket scaling in 3D views
When setting up the view for 3D figures, the Y span is doubled
incorrectly. This appears to be legacy behavior and the BoundingBox
itself accounts for the full span. Removing the 2x scaling resolves the
rocket scaling appropriately.

Fixes #788
commit 347ec93c (2020-12-19): Fix rocket scaling in 3D views

When setting up the view for 3D figures, the Y span is doubled
incorrectly. This appears to be legacy behavior and the BoundingBox
itself accounts for the full span. Removing the 2x scaling resolves the
rocket scaling appropriately.

Fixes #788
closed 2020-12-19 by wolsen