openrocket/openrocket #791: Problems with 3D rendering on Ubuntu 20.04

OZ1SEJ, 2020-10-08:
I'm running OpenRocket 15.03 using the .AppImage file on Ubuntu 20.04. The main part of the program works just fine, but when I choose 3D view of the rocket, I get a graphics error where the rocket should be: "Unable to load 3d Libraries: Profile GL2 is not available on X11GraphicsDevice[..."

Also, when I try to enter Photo Studio, I get an error dialog "NullPointerException: null" as well as a lengthy bug report.

[image]
[image]
[bug_report.txt](https://github.com/openrocket/openrocket/files/5351088/bug_report.txt)

wolsen, 2020-10-10:
I've seen this from time to time on Ubuntu when something in the graphics stack gets updated. Generally a restart of the X system or a reboot will resolve the issue.

JoePfeiffer, 2020-10-11:
tl;dr I've put up a new AppImage at https://github.com/openrocket/openrocket/releases/download/release-15.03/OpenRocket-15.03.AppImage that should fix it.  Please download it and give it a spin.

The problem is definitely an issue caused by recent upgrades to the mesa libraries -- if I downgrade  libgbm1, libegl-mesa0, and libgl1-mesa-dri to version 18.3.6-2+deb10u1 (stable), and install libllvm7,  then 3D rendering starts working; returning to 20.1.9-1 (testing, unstable) or 20.2.0-1 (experimental)  versions breaks it again. So, I've added the old versions of the libraries to the AppImage, and it works for me now.

Also note that it appears to be a bug in those libraries -- I got the error running either 15.03 or the latest unstable.  So it isn't us, and it isn't even 15.03 needing an antique version of Java.

wolsen, 2020-11-11:
As this appears to be an issue with the Debian/Ubuntu mesa libraries - the bug should be raised on [on launchpad](https://bugs.launchpad.net/ubuntu/+source/mesa)

closed 2020-11-11 by wolsen

IggyJ, 2021-11-13:
Having the same issue on Arch Linux. Using the AppImage did not fix it for me, nor did rebooting or restarting the X system.
I'm using an i7-8565U with UHD 620 integrated graphics. Running the 5.10.78-1-lts kernel.

OZ1SEJ, 2021-11-13:
@JoePfeiffer So sorry I didn't get around to try your new AppImage before, but it works exactly the same way as the one I already had installed.

@IggyJ That works! thank you so, so much! :-D