openrocket/openrocket #792: Can't clone and build unstable. Info, pls.

jimmiedave, 2020-10-10:
Hi, I've been trying to build unstable in eclipse for hours now, and I get stack traces on app startup:

> _Caused by: com.google.common.util.concurrent.ExecutionError: java.lang.NoClassDefFoundError: javax/script/ScriptEngineFactory_

I'm on Mac OS Catalina, and adoptopenjdk-11.jdk

Out of the box, I can't build a working (if unstable) jar. I assume it's a missed dependency, or a pathing configuration problem.

Can you let me know how to get a basic config up, so I can look over this, and perhaps see how I might contribute to the project? There's not really documentation on this part, and it's not clear where I'd reach out aside from filing an issue.

I'd like to help where I can.

Thanks for reading.

hcraigmiller, 2020-10-10:
Provide me with your e-mail to hcraigmiller@gmail.com and I can provide you with a current .jar file. But, I don't do Mac though.

JoePfeiffer, 2020-10-10:
That should be part of your JDK...  If you just type 'ant' from the command line to build it, and java -jar swing/build/jar/OpenRocket.jar to run it, does it work?

jimmiedave, 2020-10-10:
Thanks @hcraigmiller , I appreciate the offer, but really looking for build help more than just the working jar.

@JoePfeiffer That works fine. I'll have to look at my Eclipse configuration and see what's going on there. That'll probably take a bunch of time and head-banging, but I have at least one way to build. I appreciate your assistance.

Is there a dev list or other place where I can offer any assistance I can provide? I code in a few languages, though Java's not really one of them (OK, I've done 'hello world', but that barely counts). I enjoy debugging, write well in English, and do UX/Usability for a living (wouldn't dream of suggesting you fiddle around with UI before a next long-awaited release - not thinking of that).

Absent more chat on this, feel free to close, with my deep appreciation for the assistance.

JoePfeiffer, 2020-10-10:
I've never gotten along with IDEs at all.  I was really happy to learn I could work on this with emacs and the command line!

There is an openrocket-devel mailing list; you can subscribe at https://lists.sourceforge.net/lists/listinfo/openrocket-devel
There's also a slack channel at openrocket.slack.com

Best way to contribute is pick one of the issues on the list and dig in!

jimmiedave, 2020-10-10:
I understand where you're coming from; I worked in C with MicroEMACS on the Amiga in the long-ago '80s. I get it.

And I might've said I just need a good text editor, but in the last few years have been using Pycharm (from the JetBrains guys) for Python, and am really in love. It's not perfect, but damn, it's good.

May have to look at the community edition of IntelliJ IDEA

jimmiedave, 2020-10-11:
To put a bow on this, downloaded IntelliJ IDEA community edition, and openrocket built first time, no muss, no fuss. Eclipse is going away now...

Thanks all for your help.

closed 2020-10-11 by jimmiedave

roguex-tech, 2020-10-11:
To second the bow of @jimmiedave , I downloaded and did the exact same. No fuss, no muss nor anything else. Eclipse will never be used again by me.

JP

teyrana, 2020-10-12:
Just to add on: (unofficially)
We don't officially support or prohibit any build system, but the ones under active maintenance are ant (for the CI builds) and Intellij for development.

(As you've already discovered ;)

roguex-tech, 2020-10-12:
Adding further:  (unofficially Unofficial)
I as well as others that follow the OpenRocket Wiki would use Eclipse based on the instructions. IntelliJ just worked right out of the box. I don't post much as I prefer to do what we did way back in the day, read and tinker. Still works to this day.

As I stated on the Slack channel, if I had the experience to I would love to port this over to iPadOS. I would be 6 t under long before I could accomplish that feat. I haven't done any serious coding since the mid 90's.

I really appreciate y'alls work here. She is coming along nicely!!!

JoePfeiffer, 2020-10-13:
One of our problems is we actually have two wikis at this point -- one at http://wiki.openrocket.info/ and one at https://github.com/openrocket/openrocket/wiki

The first one only has instructions for using with eclipse (that apparently don't work), and the latter only has instructions for
building from the command line.

Should we take the eclipse instructions out of the former, and someone who has used IntelliJ put instructions for that on the latter?

For that matter, most of the rest of the Developer's Guide on the .info wiki is pretty dated...

jimmiedave, 2020-10-13:
If there are two wikis, I suggest they be combined into one, and one of the wikis decommissioned and redirected to the other (for people and search engines who've bookmarked them).

A man with a watch knows what time it is. A man with two is never sure.

If I could be of help in this, I'd be happy to do so.

JoePfeiffer, 2020-10-13:
Take a look and see if it's something you'd like to take on!  It really needs doing.

teyrana, 2020-10-15:
> If there are two wikis, I suggest they be combined into one

If you guys decide which one to continue, can help with the Intellij build instructions :)

JoePfeiffer, 2020-10-15:
When I google openrocket, the first hit is openrocket.info, and that wiki appears in the in the list of pages under there.  Based on that, I'd suggest that be the one to continue -- but it's really up to whoever wants to do the actual work!

jimmiedave, 2020-10-16:
The mediawiki version at openrocket.info looks a lot more promising atm (based on a very minimal survey of a page or two), and you've got all that google SEO juice if it comes up first - that's a good reason to work on that one.

I will work on this, to merge the wikis, tighten up the text a little as I go. Will begin with a content audit and diff (where applicable) of both.

Will gladly accept detailed Intellij build instrux - though for me it was: 1. install Intellij, 2. open the project 3. click "build".

I'm about to open another issue (better targeted to the content we're discussing, so suggest further dialogue happen there.