openrocket/openrocket #794: Freeform fin set tab length is different when opened than when saved

hcraigmiller, 2020-10-11:
Using the 10/10/2020 merge. The attached files feature three identical freeform fin sets, but the fin tabs are positioned relative to the top, middle, and bottom of the fin.

FREEFORM FIN SET

[Lotus 5.Rev.Error.Rev.01.zip](https://github.com/openrocket/openrocket/files/5360576/Lotus.5.Rev.Error.Rev.01.zip)

1. Open the fin tab for the "top" fin set, select calculate automatically, the tab length, height, and position _correctly_ show 10, 1, 1, respectively. Close the freeform fin set configuration window, then reopen the fin tab, the tab length, height, and position _correctly_ show 10, 1, 1, respectively. 

2.  Open the fin tab for the "middle" fin set, select calculate automatically, the tab length, height, and position **_incorrectly_** show 10, 1, **1**, respectively. Close the freeform fin set configuration window, then reopen the fin tab, the tab length, height, and position _correctly_ show 10, 1, **0**, respectively. 

3. Open the fin tab for the "bottom" fin set, select calculate automatically, the tab length, height, and position **_incorrectly_** show 10, 1, **1**, respectively. Close the freeform fin set configuration window, then reopen the fin tab, the tab length, height, and position _correctly_ show 10, 1, **-1**, respectively. 

Verify visually that the tab on each set of fins is correctly positioned by selecting the component freeform fin sets individually.  Save, then reopen the design file. The _tab length (**incorrect in each instance**)_, height, and position of each fin set are now:

1. Top       -  length, height, and position are **_.969_**, 1, 1
2. Middle  -  length, height, and position are **_1.969_**, 1, 0
3. Bottom -  length, height, and position are **_1.969_**, 1, -1

Again, in each instance, _**the tab length is now incorrect**_.

TRAPEZOIDAL FIN SET

[Lotus 5.Rev.Error.Rev.02.zip](https://github.com/openrocket/openrocket/files/5360591/Lotus.5.Rev.Error.Rev.02.zip)

1. Open the fin tab for the "top" fin set, select calculate automatically, the tab length, height, and position _correctly_ show 10, 1, 1, respectively. Close the freeform fin set configuration window, then reopen the fin tab, the tab length, height, and position _correctly_ show 10, 1, 1, respectively. 

2.  Open the fin tab for the "middle" fin set, select calculate automatically, the tab length, height, and position **_incorrectly_** show 10, 1, **1**, respectively. Close the freeform fin set configuration window, then reopen the fin tab, the tab length, height, and position _correctly_ show 10, 1, **0**, respectively. 

3. Open the fin tab for the "bottom" fin set, select calculate automatically, the tab length, height, and position **_incorrectly_** show 10, 1, **1**, respectively. Close the freeform fin set configuration window, then reopen the fin tab, the tab length, height, and position _correctly_ show 10, 1, **-1**, respectively. 

Verify visually that the tab on each set of fins is correctly positioned by selecting the component freeform fin sets individually.  Save, then reopen the design file. **The tab positions remain _correct_ and unchanged.**

ELIPTICAL FIN SET

[Lotus 5.Rev.Error.Rev.03.zip](https://github.com/openrocket/openrocket/files/5360600/Lotus.5.Rev.Error.Rev.03.zip)

1. Open the fin tab for the "top" fin set, select calculate automatically, the tab length, height, and position _correctly_ show 10, 1, 1, respectively. Close the freeform fin set configuration window, then reopen the fin tab, the tab length, height, and position _correctly_ show 10, 1, 1, respectively. 

2.  Open the fin tab for the "middle" fin set, select calculate automatically, the tab length, height, and position **_incorrectly_** show 10, 1, **1**, respectively. Close the freeform fin set configuration window, then reopen the fin tab, the tab length, height, and position _correctly_ show 10, 1, **0**, respectively. 

3. Open the fin tab for the "bottom" fin set, select calculate automatically, the tab length, height, and position **_incorrectly_** show 10, 1, **1**, respectively. Close the freeform fin set configuration window, then reopen the fin tab, the tab length, height, and position _correctly_ show 10, 1, **-1**, respectively. 

Verify visually that the tab on each set of fins is correctly positioned by selecting the component freeform fin sets individually.  Save, then reopen the design file. **The tab positions remain _correct_ and unchanged.**

teyrana, 2020-10-11:
Ummm, I'm not sure what's going on here... a few observations to start: 

> The attached files feature three identical freeform fin sets, 

Sure, they each have 3 fin-sets, but each set has a 3-fin-count.  and they overlap?     This is an obviously non-physical configuration, so it's hard guess what the design intent is, here.

> Open the fin tab for the "top" fin set, select calculate automatically, the tab length, height, and position correctly show 10, 1, 1, respectively.

The file shows:
```
              <tabheight>0.025400000000000006</tabheight>
                <tablength>0.02460000000000008</tablength>
                <tabposition relativeto="top">0.025399999999999923</tabposition>
```
Which corresponds to tablength=2.46 cm, tab height = 2.54 cm, and tab position = 2.54 cm from the leading edge.
The configure dialog and the side view matches these numbers.

> ... select calculate automatically, ....

This button does nothing for me....
1. What does it do for you? 
2. What do you expect it to do?

hcraigmiller, 2020-10-11:
It was not a design intent, just the easiest way to compare when I found the error. With regard to all fin types, the first error is that when you select calculate automatically, what you expect to be **immediately displayed** is:
Top: 10, 1, 1
Middle: 10, 1, **1** [incorrect]
Bottom: 10, 1, **1** [incorrect]
What is **actually immediately displayed** :
Top: 10, 1, 1
Middle: 10, 1, **0** [correct]
Bottom: 10, 1, **-1** [correct]

The second error, applicable only to freeform fin sets, is that when you save and reopen the file, the fin tabs are different, depending on relative position.  The tab length (bold number is incorrect in each instance), height, and position of each fin set are now:
Top - length, height, and position are **.969**, 1, 1
Middle - length, height, and position are **1.969**, 1, 0
Bottom - length, height, and position are **1.969**, 1, -1
Again, in each instance, the tab length is now incorrect.

hcraigmiller, 2020-10-19:
I reproduced the error using the attached file which utilizes a single set of freeform fins.
[Reaper.Error.Fin Tab.Freeform.zip](https://github.com/openrocket/openrocket/files/5405104/Reaper.Error.Fin.Tab.Freeform.zip)

Rather than trying to explain it in words, let me show you what I found.

First, open the design file, select Main fin set, then select fin tabs.

[image]

hcraigmiller, 2020-12-04:
New issue #834 created for remaining errors.

closed 2020-12-04 by hcraigmiller