openrocket/openrocket #797: Wiki content audit, freshen and merge

jimmiedave, 2020-10-16:
Regarding updating the Wikis for openrocket.

At issue: openrocket has two documentation wikis. One on this site (linked above at "Wiki") and another linked from openrocket.info. Both wikis have a certain amount of out-of-date information, and there should be a single, up-to-date definitive source for doc and info.

Proposed: 
- Retain the openrocket.info wiki because it is first-ranked in google for info on openrocket - no need to SEO it to get it higher.
- If possible, provide a permanent redirect from the GitHub wiki to the openrocket.info wiki, so that peoples' legacy bookmarks and search engine corpi get updated.
- Audit and freshen the wiki content, removing any ROT (redundant, outdated, trivial content) and cruft, posting current information.
- Anticipate a future release by planning for new content.

Needs:
- Style guidance: orthodox spelling of "openrocket" or "Open Rocket" or "OpenRocket" or "Openrocket", etc. as well as other fine points on the language around the project
- Any appropriate graphics. Logo file would be a great place to start.

Open to your thoughts.

jimmiedave, 2020-10-18:
No pictures at all on openrocket.info wiki?

Is this a server misconfiguration, or have they all been wiped?

JoePfeiffer, 2020-10-20:
Is it possible it's just that no one has put any pictures on it?

jimmiedave, 2020-10-20:
Nope - there are filenames, URLs and publish dates - the first one in the "basic rocket design" page is shown as 2013, so I figure they've gotten lost during a move, or perhaps the Wiki expects to use ImageMagick or GD something and it's not configured.

Main window image URL:

http://wiki.openrocket.info/images/1/1f/Main_window.png

Who is the server admin for openrocket.info?

neilweinstock, 2020-10-22:
When I used MediaWiki in the past, pictures were a pain.  I don't know if it's better now.  I also generally prefer better support for hierarchy.

I was hoping the Github wiki would become the canonical documentation source for OR.

Whichever direction it goes, a final decision would be good, because I'm not going to start working on docs until it is made.

jimmiedave, 2020-11-01:
Note in to Sampo (who apparently controls the Wiki site) requesting a fix on images, offering my assistance if he'd like.
---
@neilweinstock do you think that perhaps the github wiki would be the place for development documentation, and the openrocket.info the place for user doc? That would seem like a reasonable decision to make, and we can point one wiki at the other when the visitor wants user or developer doc.

neilweinstock, 2020-11-02:
That is possible; at least there would be clear separation of function between the two.

I don't look forward to doing a lot of content entry on MediaWiki due to the way it manages images, but I would live with it.  Who makes final decision?

JoePfeiffer, 2020-11-03:
The more I think about it, the more I like your idea of having a development wiki on gihub, and a user wiki on openrocket.info.  As Neil says, it'll provide a good separation.

jimmiedave, 2020-11-13:
Trying again to reach Sampo about server misconfiguration. if anyone else has keys to the server that hosts the wiki, please reach out so we can get the config fixed.

neilweinstock, 2020-11-13:
> Trying again to reach Sampo about server misconfiguration. if anyone else has keys to the server that hosts the wiki, please reach out so we can get the config fixed.

I am also curious about how old a version of Mediawiki that is, and whether it can be brought more nearly up to date.

I looked through the content, and it seems like it's going to be challenging to modify it without in many cases restarting from scratch (although it should be possible to re-use significant chunks of text).  Furthermore, I feel like we need to keep 15.03 docs separate from new docs, *at least* until release, and possibly longer.  What's a good way to do that?

Finally, if anyone is fluent in Mediawiki, I could use some input on how to work with a flat structure.  For instance, right now the FAQ is a single page, but it would make much more sense to have each FAQ item on its own page.  What's the right way to lump them together, is it categories?  Likewise, user guide pages should be lumped together, reference pages should, etc.  Normally I prefer wikis that support hierarchical structure to make this sort of thing easier and/or clearer, but I'm sure there are conventions on how to do such things on Mediawiki.

jimmiedave, 2020-11-13:
Removed empty end-user guide pages from GitHub wiki, added minor text tweaks about this being the developer wiki, user wiki on openrocket.info

Left more technical end-user doc (scripting, solving java probs) on the GitHub wiki. As OpenRocket.info takes more shape, the separation can be a bit more distinct.

jimmiedave, 2020-11-13:
> I am also curious about how old a version of Mediawiki that is, and whether it can be brought more nearly up to date.

I looked at the version info a couple weeks ago, and IIRC it's the latest. It may be set to autoupdate - and in fact, that could've moved the wiki away from a not-updated server configuration for image handling.

neilweinstock, 2020-11-13:
Ah, interesting.  I couldn't figure out where to find the version.  I noted
that it doesn't have the WYSIWYG editor, perhaps that is an extension or a
config option.

jimmiedave, 2020-11-13:
Clobbered refs to SourceForge and fixed broken links on Introduction page.

Missing images sucks, but won't prevent work on the text.

jimmiedave, 2020-11-18:
Need assistance from a Java SME: The [install page](http://wiki.openrocket.info/Downloading_%26_Installing) is wildly out of date as regards Java. Can I get someone who knows what's what to give a look, suggest or perform edits?

1. Does 15.03 not work with Java > 8?
2. Does the Java uninstaller not detect OpenJRE? Seems like it.
3. OpenJRE: why don't we document this?

This page needs a major look-over from someone who knows what (s)he's looking at.

---

Removed SourceForge links from this page this evening.

jimmiedave, 2020-11-18:
Images are truly broken, not just missing. Uploaded a new copy of the logo, just as missing as the rest of the images, so that says the site is misconfigured.

JoePfeiffer, 2020-11-18:
> Need assistance from a Java SME: The [install page](http://wiki.openrocket.info/Downloading_%26_Installing) is wildly out of date as regards Java. Can I get someone who knows what's what to give a look, suggest or perform edits?
> 
> 1. Does 15.03 not work with Java > 8?
This is correct.

> 2. Does the Java uninstaller not detect OpenJRE? Seems like it.
> 3. OpenJRE: why don't we document this?
Don't know the answers to these, sorry.

> This page needs a major look-over from someone who knows what (s)he's looking at.
The information about the Linux AppImage is, to the best of my knowledge, correct (I wrote it).
From what Bdale Garbee has said, I believe OR either has been or is about to be removed from the Debian repositories since the required version of java is too old.

neilweinstock, 2020-11-18:
>    1. Does the Java uninstaller not detect OpenJRE? Seems like it.
>    2. OpenJRE: why don't we document this?

There's also nothing there about the packaged installers.

Again I am moved to ask: are we going to keep docs for the new version
separate from the old?  If so, I would be disinclined to spend too much
effort on the old docs at this point.

jimmiedave, 2020-11-18:
@neilweinstock Separate docs are fine, though:

1. Until a new release, the old are all people have. IIUC, the "18" in version number means this has been kicking around since 2018. If I'm mistaken, please correct me. So if a new version is coming "real soon now" for a long time, it behooves us not to leave current version users without doc.

2. The fastest start to a new version's doc would, arguably, be to copy (good) docs from the old version and update them, as the new functionality doesn't represent a sea-change to the app (AFAICT). So improvement to the old doc, prior to starting the new, improves the ease of updating for the new.

Certainly the Java version and Install doc will differ.

As always, I'm open to being convinced of something different.

jimmiedave, 2020-11-19:
Reached out a third time to Sampo / @plaa - this time with a comment on a long-closed issue. Willing to take nearly anything that gets us to a working wiki and a maintainable site at this point.

plaa, 2020-11-19:
Hi,

Sorry for the delay - each time I read @jimmiedave's email I was unable to react immediately and then forgot.  I believe the hosting provider did some Apache configuration updates that broke it (or a version upgrade I recently did).  Images are fixed now.

I host a couple of MediaWikis for various purposes.  It's a personal account hosting several sites, so unfortunately I can't provide access to anyone else.  I can continue hosting it (it doesn't cost anything extra) but don't object to transferring it to a ready-hosted (i.e. more reliable) option.  I'll leave that decision up to you.

As for the name, I prefer the spelling OpenRocket, but am not adamant about that either.

jimmiedave, 2020-11-19:
Thanks very much @plaa I really do appreciate this; it'll help us move forward.

I'll reach back out about hosting once the Wiki's a little better updated. If we'd change hosting, ideally there would be a way to have the project govern it, rather than putting the site again in the hands of a single person who may be busy or out of contact at any given time.

I'll try and figure out how other projects do this.

jimmiedave, 2020-11-20:
Updated (slightly) install instructions. @neilweinstock you could have a look at the canary-yellow box at the top, recommending the packaged installers for now: [(http://wiki.openrocket.info/Downloading_%26_Installing)]

Updated macOS terminology to the latest; preserved "Mac OS X" for earlier versions where appropriate.

Removed link to scary "UNINSTALL JAVA" page, which the OpenRocket install page suggested users use to determine the Java version they have. Left command-line version detection intact.

jimmiedave, 2020-12-03:
Update to semantics on Wiki section 1.
Updated images and text on Wiki section 3 - there's a lot to do here. Fortunately, there are generally good semantics here.

JoePfeiffer, 2020-12-03:
What do you mean by "semantics", in this context?

jimmiedave, 2020-12-03:
> What do you mean by "semantics", in this context?

As I edit the text, one thing I see is use of the `<b>` tag for bolding, (and others). There's nothing specifically wrong with that - it's the way we used to do things - but it carries no semantic context.

If, f.e. someone is reading with a screen reader, or gets a snippet of the page by asking 'OK Google' a question that returns this as an answer, there's no way to "bold" computer-spoken voice in the response. If instead of `<b>` one uses `<strong>`, the web browser will bold the text, and the computer voice can modulate its speech to indicate the word has extra importance. 

The `<em>` tag works similarly with italics. Other semantic improvements might be reviewing document organization with heading tags and paragraphs, vs. tags in <div>s that might just be formatted to _look like_ headers, but which don't carry the semantic indication that they head a section. These indicators are of particular importance to SEO.

Working semantically with the content improves its accessibility, its searchability/findability, and its readiness for future technologies we don't yet know of: it improves the "aboutness" of the content, vs. simply its formatting.

JoePfeiffer, 2020-12-03:
Ah, got it.  Semantic rather than non-semantic html tags.  Thanks.

jimmiedave, 2020-12-13:
Updated "Getting Started" page images to 15.03 up through "Scale Window". "Preferences" is very different, and will need some new sections written - will ask if I need assistance.

Logging images for reuse in a v.18 (or whatever) release - can recapture as we get close to that release. Expect to use MediaWiki namespacing to keep v.15 and v.18 docs separate.

Also: working on getting local rocketry organization (an NAR section) to take on stewardship of the Info site/wiki. This will ensure that domain names and certs are not in the ownership of a single person (a single point of failure at domain/cert renewal time: an expired card, or some bad luck are all it takes to cause a problem).

Likewise, trying to provide a couple admins so again, not a single point of failure. Quite open to comments on this idea, and wondering what approval will be needed.

Site stewardship is an effort at this point, not a guaranteed thing happening.

jimmiedave, 2020-12-13:
Updated "Getting Started" page to 15.03 up through Preferences. The Preferences sections are largely new writing. If someone wants to check behind me, that'd be a good idea. Wrapping work for the weekend here.

neilweinstock, 2020-12-22:
@jimmiedave Sorry I've been AWOL on this for a while, will try to get back into it and help out.  Should we add a slack channel for docs?  Might be more manageable than a single comment thread here; there are multiple issues I'd like to discuss.  Are you on the OR Slack?

In the meantime:
> Updated (slightly) install instructions. @neilweinstock you could have a look at the canary-yellow box at the top, recommending the packaged installers for now: [(http://wiki.openrocket.info/Downloading_%26_Installing)]

Two things here:
1) I would prefer to focus primarily on the installer packages, and deprecate the JAR files completely.  In other words, turn that yellow box into the majority of the page.  This requires a full rewrite.
2) For the packaged installers, you've linked to a release posting by Wes back in 2015.  Were all those packaged installers put there back then?  If so, holy crap, neither I nor (I assume) just about anyone else realized it.  Or were they added after the fact?  If so, how can you even tell?

neilweinstock, 2020-12-22:
Hard-coded mapping of pages to user guide sections leads to ugliness like this:
[image]

However, I'm not fluent enough in MediaWiki to know a better way to do this sort of thing.  My insatiable desire to impose some sort of structure on wiki pages inevitably leads me to conflict with MediaWiki.

jimmiedave, 2020-12-22:
> Hard-coded mapping of pages to user guide sections leads to ugliness like[...]

Have a look here:. I think this is what you were trying to get to. Let me know if this is it, and I'll do the rest of the pages and contents to match:

http://wiki.openrocket.info/Downloading_%26_Installing

jimmiedave, 2020-12-22:
>/.../Are you on the OR Slack?

No, but if you want to invite me, we can move discussion there as you suggest.

>    /.../In other words, turn that yellow box into the majority of the page.  This requires a full rewrite.

Fine with me, though there's other work a little ahead of it. Will revisit after finishing re-screenshotting/fixing Download & Install. If you want to dive in, that'd be fine - just let me know so I don't clobber your changes.

>    /.../Were all those packaged installers put there back then?  /.../ Or were they added after the fact?  If so, how can you even tell?

Since these are github releases, looks like it from the dates. Am going to guess that's about the time this project came to github, since most of the releases are the same day in November 2015. The tech doc is from a version 13, and an Oct. 2015 release date, IIRC. The Releases page is not in the Wayback to compare.

Will get back into a server move after Christmas, see if I can get our NAR section to approve (seems likely). Once there, we can namespace these docs for v. 15 and do a copy or new area for upcoming version.....21?

jimmiedave, 2021-01-20:
@plaa If you're available for a chat/call/email thread/something sometime soon, I'd like to begin migration of the OpenRocket.info site to a new host.

My NAR section: SoAR (Southern Area Rocketry), NAR section #571, has offered to pick up hosting and domain name registration, and other associated costs for hosting. This arrangement will provide a couple of admins for the site, and hosting and domain name registration will be in the hands of the section, not of an individual. (I could've afforded to pick all this up myself, but in Covid-world, all it takes is some bad luck for me to be out of commission for a while or forever...)

We expect the site to remain largely the same for the time being, minor changes include continuing to update the Wiki, updating doc as needed for future release, potentially hosting release packages, and working hard to optimize SEO so that the site becomes the go-to for correct and timely info on the latest OpenRocket.

I'd like to get the Wiki and site we have presently stabilized on a new host before moving the domain - the domain transition can happen all at once (registration and new host).

Happy to answer any questions you may have.

plaa, 2021-01-27:
@jimmiedave Hi, sorry for the slow response.  Feel free to ping me again if you're missing an answer.

I think the largest bottleneck is the lack of site admins (i.e. me).  This could be handled by pointing the domain to a wiki hosted elsewhere.  (Currently it's hosted under my personal account which I cannot share.)

While I'm perfectly fine with handling the domain costs, I understand that having the domain ownership with a team of people is preferable.  Therefore I'm fine with transferring the domain ownership or adding additional people as admins.

I would first like to hear from the other active members what they think is the best solution.  I can transfer full ownership to SoAR, or I could create an organization in Gandi and add more people as admins there.

PS.  Here's an OpenRocket Slack invite link.  Should it be put on the home page as well?
https://join.slack.com/t/openrocket/shared_invite/zt-dh0wtpc4-WmkSK1ysqAOqHa6eFN7zgA

jimmiedave, 2021-01-28:
I too would like to hear from others.

I'd like to be able to maintain and configure the wiki (for example, enabling WYSIWYG editing, better image tools, configuring tagging or namespacing so we could keep versioned docs), and I don't feel comfortable asking your time and engagement on it, much less like I ought to be mucking around on your own server.

But I don't want to make this "my show" either - I haven't close the time to do so, were I so inclined. Would like to improve things a bit, update doc, then take on the periodic care and feeding, bring along another admin, removing single points of failure.

jimmiedave, 2021-03-11:
OK, after a little more than a month with no comment, and a SoAR meeting tomorrow night, I believe **I'm going to have to close out the offer to take over hosting**.

For the record, things that will need doing, that someone who's an administrator has to do:

- Adjust the menus and content on the web site so there's a single, correct place to get packaged installers of 15.03 until there's a new release available.
- Do the same for said new release.
- Remove or archive deleted cruft pages from the wiki
- Make the path from web site front page to user guide explicit and easy to find.
- Add WYSIWYG editor extension to MediaWiki to make authoring easier
- Create and maintain a method of separating 15.03 content from _{shiny new release}_ content, where someone with 15.03 can still find doc. This might use namespacing, or tagging, but seems it will require an admin be involved, and will require an investment of time.
- Fix whatever's causing MediaWiki to space out and not let _{user}_ actuate the tabs in preferences.

Also:

- Not Openrocket.info, but related: close off the SourceForge account (or remove old stuff) if it's not hosting current source and executable: it's got a high position in SEO rankings, and will serve to confuse people, especially after _{shiny new release}_ shows up. The SourceForge presence, if it's to be kept (yah, mailing list...) should redirect back to OpenRocket.info in all cases.
- Adjust any needed OpenRocket.info content for the prior bullet on SourceForge close-off.

I've done quite a bit of work on getting the Wikis (both) current, and am close to the limit of what I can do for now. More on the latest changes in the next post.

**I'm going to step away from Wiki updating for now.** Here are what I see as the TODOs on the OpenRocket.info wiki (these are larger tranches of work, not sweeping up in the corners):
- Write an article on Rocket Analysis to fill out its chapter - no idea what that's for in this case, I know nothing about the domain.
- Update **Appendix B: Components** with current screengrabs and componentry from 15.03
- Carve off and update the 15.03 content for _{shiny new release}_ whenever.
- Update the feature comparison with RockSim to the latest - they've got a new release out this year, IIUC. Maybe give Tim Milligan a chance to comment on the comparison. Add in info on SpaceCAD, NASA RocketModeler, ThrustCurve - could be generally useful for anyone trying to understand what's out there for the various purposes.

I can provide, for the medium-term foreseeable future, a list of the screengrabs I took for the update to current - this may be of use to anyone creating grabs for _{shiny new release}_.

I wish whoever will administer and update the best of luck!

jimmiedave, 2021-03-11:
Recent OpenRocket Wiki changes:

- [Main Page](http://wiki.openrocket.info) primarily now hosts the user guide.
    - Didn't delete any content (other than empty sections where the text muses about what should go there, instead of providing content); it's now in the appendix to the user guide, or on the Developer wiki, if it's developer content.
- Hard split of Developer and End-User wikis. 
    - [End-User Documentation Wiki](http://wiki.openrocket.info/) (on OpenRocket.info)
    - [Developer Documentation Wiki](https://github.com/openrocket/openrocket/wiki) (on GitHub)
- Each Wiki points at the other for their respective Developer or End-User purpose.
- This should improve the reliability and findability of the content: there's one place to find the definitive document, one place for Google to locate and index it (so it becomes the definitive page on the topic), and also only one place to have to edit it.
- I populated and formatted the empty Developer Wiki sections with the content moving over from the openrocket.info site.
- I linked a couple Developer Wiki pages back to the End User wiki (visible in the ToC on Developer Wiki) because the technical detail in those primarily-end-user pages might be of interest to developers.
- Greatly fleshed out the Advanced Rocket Design article adding techniques that use the 15.03 capabilities, and added images. The entire Wiki could use more interesting images.
- Nearly all screenshots updated to 15.03 screen - if not all of them, except for **Appendix B: Components**.
- There should be no more references to SourceForge in the End-User Wiki - except perhaps to the mailing list.

I hope someone finds the work useful, and a more up-to-date reference for documenting _{shiny new release}_