openrocket/openrocket #799: Simulation crashes openjdk-13

JoePfeiffer, 2020-10-19:
The attached ork file is causing JRE exceptions with several recent versions of Java.  In all cases, the fatal error is happening in

Current CompileTask:
C2:  11710 5947       4       net.sf.openrocket.simulation.listeners.SimulationListenerHelper::firePostAerodynamicCalculation (119 bytes)
(there are differences in the other information on the line, but all of them point to firePostAerodynamicCalculation)

Note: this file was shared to The Rocketry Forum in the thread
https://www.rocketryforum.com/threads/open-rocket-doubt-flight-event-exception.162328/
The author appears to recognize it's a completely impractical design, but it shouldn't be crashing the JRE!

I'm attaching the original .ork (filename extension changed to .txt to pacify github's uploader), and all the error dumps I've got.
[rocket-2020-10-17.txt](https://github.com/openrocket/openrocket/files/5405321/rocket-2020-10-17.txt)
[hs_err_pid1744172.log](https://github.com/openrocket/openrocket/files/5405324/hs_err_pid1744172.log)
[hs_err_pid1744540.log](https://github.com/openrocket/openrocket/files/5405325/hs_err_pid1744540.log)
[hs_err_pid1745334.log](https://github.com/openrocket/openrocket/files/5405326/hs_err_pid1745334.log)
[replay_pid1744540.log](https://github.com/openrocket/openrocket/files/5405334/replay_pid1744540.log)

ore13, 2020-10-23:
I've had the same error, but occurring with some of the example rockets. I've attached the error dumps I got here too. fatal error is still happening in the firePostAerodynamicCalculation compile task.

A simple model rocket: 
[hs_err_pid16748.log](https://github.com/openrocket/openrocket/files/5427467/hs_err_pid16748.log)
[replay_pid16748.log](https://github.com/openrocket/openrocket/files/5427468/replay_pid16748.log)

3 stage example rocket:
[hs_err_pid15296.log](https://github.com/openrocket/openrocket/files/5427470/hs_err_pid15296.log)
[replay_pid15296.log](https://github.com/openrocket/openrocket/files/5427471/replay_pid15296.log)

Typically takes 2 simulationOption changes (then run) to crash when using the 3 stage example rocket, and  4 - 10 simulationOption changes to crash the simple model rocket.

JoePfeiffer, 2020-10-24:
This appears to be a bug in the JVM's runtime loop optimizer, similar to https://bugs.openjdk.java.net/browse/JDK-8229219
That bug, and several other similar ones I've found, are marked "resolved" but the JVM sources I downloaded and compiled still show the behavior.  I'll attach the crash output from the debug version of the JVM; I've wasted a little time trying to make changes to SimulationListenerHelper:firePostAerodynamicCalculation() that might avoid it, but I don't know the JVM anything resembling well enough to do a real job understanding exactly what we're doing to trigger the bug and my guesses haven't helped.  If someone else can see a workaround, go for it.
[openrocket.log](https://github.com/openrocket/openrocket/files/5433044/openrocket.log)

JoePfeiffer, 2020-10-26:
FWIW, java 15 (adoptopenjdk-15-hotspot) seems to simulate the rocket I tried successfully.  ore13, would you be able to install that and see if it clears up your crashes as well?

ore13, 2020-10-27:
I've been using Java 15 to run my simulations. It seems to work fine. I've probably run about ~500 simulations on various rockets, java 15 hasn't crashed on me yet

JoePfeiffer, 2020-10-27:
Also, it looks like they're backporting the fix to at least -13 and -14.  So I'm going to mark this closed...

closed 2020-10-27 by JoePfeiffer