openrocket/openrocket #803: Simulation launch configuration settings not saved properly

takd, 2020-10-26:
If I run a simulation with custom launch and wind direction, it runs correctly, and the results are saved properly. Hovewer, after saving the .ork file, relaunching the program and reloading the previously saved file, some _parameters_ are reset to their default value, and re-reunning these simulations without changing anything changes their results. The values of the settings are not present in the XML file inside the ork file either.

Steps to reproduce:
- Open any example file
- Change the settings of a simulation to arbitrary values, e.g.
[image]

- Run simulation(s)
- Save the ORK file
- Exit the program
- Relaunch the program, load the previously opened file
- Some settings are reset, e.g. continuing the previous example:

[image]

This also affects copy-pasting the simulations within the program.

The whole issue is really frustrating to me, since I am trying to create a lot of different simulations with different wind parameters.

Tested against latest stable release and the latest unstable. Platform specs: (Manjaro) Linux; ```openjdk version "1.8.0_265"; OpenJDK Runtime Environment (build 1.8.0_265-b01); OpenJDK 64-Bit Server VM (build 25.265-b01, mixed mode)```

JoePfeiffer, 2020-10-27:
Correct.  For better or worse, simulation options are not regarded as part of the simulation, and aren't saved.

I'm thinking for a future release it would be a good idea to have named, saveable simulation settings -- see #673 (my idea was more for someone to have different options for different launch fields, but it would work equally well for use cases like you're describing).

takd, 2020-10-27:
Yes, #673 would be a great thing.

I am a bit surprised that this isn't a bug, but rather a feature not yet implemented; what is the difference between the options saved and not saved? To me it seems that all are equally important for a simulation.

JoePfeiffer, 2020-10-27:
I haven't done any sort of exhaustive analysis, but my impression is that everything related to the rocket itself (and for this discussion that includes the motor) is saved; everything external is not.  The choice was made before my time.

Rather than calling it either a feature or a bug, I regard it as a missing feature that needs to be added (but not on this release; it's already been 'way too long between releases).

closed 2020-11-03 by JoePfeiffer