openrocket/openrocket #807: Downloading motor files fails without diagnostic

JoePfeiffer, 2020-11-03:
Several of the motor files at thrustcurve.org are failing to download (the vast majority are succeeding).  The only diagnostic is a message similar to " [java] 	Error downloading RockSim for motorID=782".  Some hint as to what's wrong is needed...

hcraigmiller, 2020-11-04:
Perhaps a conversation with John Coker on TRF would be helpful.
[image]

JoePfeiffer, 2020-11-05:
The error message turns out to be a red herring -- in the case where there is no data for the motor in the specified download format (ie RASP or RockSim), attempting to put null data in the list of curves for the motor throws an error.  Actually harmless, but alarming and misleading.

commit 6582b24b (2020-11-07): Merge pull request #809 from JoePfeiffer/fix-807

get rid of error message when no data for motor in specified format (fixes #807)
closed 2020-11-07 by JoePfeiffer