openrocket/openrocket #810: "Save as..." followed selecting files of type "All Files" results in "Uncaught exception"

hcraigmiller, 2020-11-05:
Open a simple model rocket example, select "Save as...", select files of type "All Files". This results in an "Uncaught exception", but, when you close the uncaught exception window, all of the files are displayed and the programs appears to function correctly.

teyrana, 2020-11-07:
resultant stack trace from error log:
 
```
---------- Exception stack trace ----------
java.lang.ClassCastException: class javax.swing.plaf.basic.BasicFileChooserUI$AcceptAllFileFilter cannot be cast to class net.sf.openrocket.gui.util.SimpleFileFilter (javax.swing.plaf.basic.BasicFileChooserUI$AcceptAllFileFilter is in module java.desktop of loader 'bootstrap'; net.sf.openrocket.gui.util.SimpleFileFilter is in unnamed module of loader 'app')
	at net.sf.openrocket.gui.main.RememberFilenamePropertyListener.propertyChange(SaveAsFileChooser.java:89)
	at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:334)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
	at java.desktop/java.awt.Component.firePropertyChange(Component.java:8728)
	at java.desktop/javax.swing.JFileChooser.setFileFilter(JFileChooser.java:1461)
	at java.desktop/com.sun.java.swing.plaf.windows.WindowsFileChooserUI$FilterComboBoxModel.setSelectedItem(WindowsFileChooserUI.java:1227)
	at java.desktop/javax.swing.JComboBox.setSelectedItem(JComboBox.java:579)
	at java.desktop/javax.swing.JComboBox.setSelectedIndex(JComboBox.java:624)
	at java.desktop/javax.swing.plaf.basic.BasicComboBoxUI.selectPreviousPossibleValue(BasicComboBoxUI.java:1193)
	at java.desktop/javax.swing.plaf.basic.BasicComboBoxUI$Actions.actionPerformed(BasicComboBoxUI.java:1620)
	at java.desktop/javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1810)
	at java.desktop/javax.swing.JComponent.processKeyBinding(JComponent.java:2900)
	at java.desktop/javax.swing.JComboBox.processKeyBinding(JComboBox.java:1431)
	at java.desktop/javax.swing.JComponent.processKeyBindings(JComponent.java:2962)
	at java.desktop/javax.swing.JComponent.processKeyEvent(JComponent.java:2862)
	at java.desktop/javax.swing.JComboBox.processKeyEvent(JComboBox.java:1423)
	at java.desktop/java.awt.Component.processEvent(Component.java:6409)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5008)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
	at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1950)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:870)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1139)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:1009)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:835)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117)
	at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
	at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at java.desktop/java.awt.Dialog.show(Dialog.java:1070)
	at java.desktop/javax.swing.JFileChooser.showDialog(JFileChooser.java:756)
	at java.desktop/javax.swing.JFileChooser.showSaveDialog(JFileChooser.java:678)
	at net.sf.openrocket.gui.main.BasicFrame.saveAsAction(BasicFrame.java:1437)
	at net.sf.openrocket.gui.main.BasicFrame.access$800(BasicFrame.java:118)
	at net.sf.openrocket.gui.main.BasicFrame$8.actionPerformed(BasicFrame.java:494)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
 . . .

```

JoePfeiffer, 2020-12-01:
Is your commit ready for a PR?

commit e4683a84 (2020-12-01): Merge pull request #812 from teyrana/fix/810/save-as-all-exception

[fixes #810] Selecting 'Save As...' > 'All Files' no longer throws an exception
commit eceaff64 (2020-12-01): [fixes #810] Selecting 'Save As...' > 'All Files' no longer throws an unneccessary exception
closed 2020-12-01 by JoePfeiffer