openrocket/openrocket #816: Can't click on rail button in side view to select

neilweinstock, 2020-11-08:
Just noticed this on my Mac.  If I click on a rail button in the side view, it doesn't select.  Every other component type seems to work normally.

wolsen, 2020-12-13:
I was able to recreate this at some point in the past, but I'm currently unable to recreate it off the tip branch. Can you try from latest tip? I've uploaded a jar file for you [here](https://www.dropbox.com/s/zzl5ca37h59u9l0/OpenRocket-unstable-05d23ce5.jar?dl=0).

neilweinstock, 2020-12-13:
I just built unstable about 30 minutes ago, and the problem is still there on my Mac.

wolsen, 2020-12-17:
Okay, I see it as well. For some reason I was focused on 3d view and not side view - apologies.

I believe its caused by the way the code selects which component is selected and the way it draws the shapes. I believe you'll find that if you rotate the view a bit and select the railbutton from the top (i.e. in the top flange of the button) then it will select the component. However, selecting the railbutton directly from the side will not work as the click needs to be within the top flange.

neilweinstock, 2020-12-17:
Yes, confirm that it sees clicks on the top flange.  That is certainly not intuitive, especially since the default position of the button is on top of the body tube, where there is no clickable area exposed.

Is it fixable?

wolsen, 2020-12-17:
> Yes, confirm that it sees clicks on the top flange. That is certainly not intuitive, especially since the default position of the button is on top of the body tube, where there is no clickable area exposed.
> 
> Is it fixable?

I believe so, but will take some thinking about how the selection code works along with how the drawing code works. I don't think it will be a trivial change.

SiboVG, 2021-06-26:
I may have found the problem. How the RailButton figure is built up in the side view is as follows: the rail button shape itself is basically 3 cylinders stacked on top of each other. However, in the Side view, this cylindrical 3d view is created by using ellipses and lines. This can be best seen if you turn the rotate the view a bit. Each cylinder top and bottom face is just an ellipse. These ellipses are then connected using straight lines (these are kind of the side faces of the 3D cylinder).

The problem why you can't select the rail button, is because what you see is not one shape. By this I mean that if are at 0° side view and try to select the middle part of the rail button, you aren't actually selecting any shapes, because to the left of the selection point and to the right of it, you just have a 2D line and to the top and bottom of it, you have a flattened 2D ellipse. In between those 2D shapes, there is just emptiness.

[image]

The reason why you can select the flanges if you rotate the view a bit, is because you are then clicking on one of the 2D ellipses. So in that case, you ARE selecting a shape. Also note that if you try to select anything else than the ellipses (e.g. one of the "side faces"), it won't work because then you are back in the first situation, where you aren't selecting a shape, but just emptiness.

I have not yet found a solution for this problem, but will look further into it. One possible solution I'm thinking of implementing is adding 3 invisible rectangles that will kinda act as the side faces of the 3 cylinders. These would need to be invisible, because the top and bottom of the rectangle can't be shown, otherwise you will not have a correct "3D" view when you rotate the view.

For anyone wondering, the side view of the shape is built in swing/src/net/sf/openrocket/gui/rocketfigure/RailButtonShapes.java.

hcraigmiller, 2021-06-26:
This comment is intended to describe the current functioning; I will leave it to @neilweinstock to comment on whether the function is correct.

In the 3D views (and the back view), if you can see a rail button, you can successfully click on it. In the side view, the follow describes the functionality on rotation:
- 5 degrees: Functions
- 10 degrees: Functions
- 15 degrees: Functions
- 20 degrees: Functions
- 25 degrees: Functions if you click an area sticking out past the airframe
- 30 degrees: Functions if you click an area sticking out past the airframe
- 35 degrees: Non-functional (the entire rail button is in from of the airframe)
From 35 degrees to 90 degrees, it does not work (the entire rail button is in from of the airframe).

The functionality is the same between 30 and 150 degrees (when the rail button _is in front of the airframe_) and between 210 and 330 degrees (when the rail button _is behind the airframe_), you cannot click on the rail button. This is the same regardless of zoom percentage, and regardless of rail button size.

Compared to a mass component (radially placed outside the airframe), such a component can be clicked in the side view, regardless of whether above or below, or in front of or behind the airframe.

[OS Windows 10 Pro service pack 120.2212.2020.0 with Java version 11.0.9 2020-10-20 LTS]

SiboVG, 2021-06-26:
It seems that this problem sometimes occurs. Tried quitting and reopening OR a couple of times and sometimes the problem that you mentioned arises, sometimes it doesn't. Will look after it.

SiboVG, 2021-06-26:
Hm it seems like the launch lug also has this problem (also not always, but only in certain instances of OR). @hcraigmiller could you check if you also experienced this problem?

SiboVG, 2021-06-26:
Just compiled the last unstable commit to check if I could recreate the problem here and it was the case. How I recreated it: just opened OR, made a new file, added body tube and launch lug and checked if I could select the lug at e.g. a view rotation of 88°. If I could, then I opened again a new file, did the same, body tube + launch lug, tested if I could select it, and again... After about 7 tries I experienced the same problem with the launch lug (mind you, this was done with the latest unstable release, so it is a problem not related to this PR).

So I guess that this PR does solve the rail button problem, because it worked all the times that the launch lug select also worked and when the rail button select did not work, the launch lug also didn't work. So this must be a new, different issue.

hcraigmiller, 2021-06-27:
Using the 06/26/2021 merge, the **launch lug** _works correctly_ at all angles of rotation in the side and back view, and if you can see the launch launch lug in 3D views.

The **rail button** _still acts as stated in my above comment_.

[OS Windows 10 Pro service pack 120.2212.2020.0 with Java version 11.0.9 2020-10-20 LTS]

SiboVG, 2021-06-27:
And did you try repeatedly opening a new file and testing it again, like I stated?

PR #961 (2021-07-17): [fixes #816] Fix select rail button side view
This PR fixes #816 where it was not possible to select the rail button from the side view.

The problem was the following:
How the RailButton figure is built up in the side view is as follows: the rail button shape itself is basically 3 cylinders stacked on top of each other. However, in the side view, this cylindrical 3D view is created by using ellipses and lines. This can be best seen if you rotate the view a bit. Each cylinder's top and bottom face is just an ellipse. These ellipses are then connected using straight lines (these are kind of the [... cut]
commit 5427edb3 (2021-07-17): [fixes #816] Fix select rail button side view
closed 2021-07-17 by wolsen