openrocket/openrocket #817: New feature request: Parachute spill hole option

hcraigmiller, 2020-11-10:
The Rocketry Forum users are requesting a parachute spill hole option.

JoePfeiffer, 2020-11-10:
Seems like a good idea (for a future release), but do we have any models of the effect of a spill hole?

hcraigmiller, 2020-11-10:
Typically, spill holes are located at the apex, and are between 10% and 30% of the parachute’s diameter. Because the spill reduce/eliminate/control uneven air spillage around the parachute edges – evidenced by a rocket spinning or oscillating back and forth on descent – the undesired forces are not generated around the edges. Some believe that a spill hole will reduce the efficiency of a well-balanced parachute, so you need to compensate by increasing the size of the parachute for a given rocket if you choose to use one. Others believe that, In most cases, you will see an improved descent rate because the loss of efficiency due to the spill hole is less than the inefficiencies of the chute’s imbalances (same or smaller chute with a spill hole has a similar descent rate as no spill hole). The standard for Fruity Chutes, one of the best on the market, is a spill hole with a diameter 20% of that of the canopy (a spill hole that is about 3% of the canopy surface area).

I will try to find some actual computational information.

JoePfeiffer, 2020-11-10:
It's the analytical modeling I was asking about.

hcraigmiller, 2020-11-10:
I'm still looking for the analytical modeling calculations... In the mean time, the articles that I have located seem to imply that you simply use the surface area of the canopy (for a flat parachute: PI*Rcanopy^2 - PI*Rspillhole^2) as a replacement for that data in the flat parachute computations.

What I think I'm really looking for is how a spill hole affects the computation of the parachute Cd.

JoePfeiffer, 2020-11-10:
I didn't realize until reading the thread over on TRF that shroud line length isn't taken into account in the parachute calculations.  I think revisiting the parachute code is probably something to put on the "right after the next release" list.

neilweinstock, 2020-11-11:
I think an analytical model of the effect of shroud length on parachute Cd is another unknown right now.  It would be nice to have, but I'm not sure it rises to the level of "fix right after release".

JoePfeiffer, 2020-11-12:
Since people use our descent rate in determining a safe flight, it seems relatively import to me.  But we should devote some thought (after next release!) to which of the feature requests are most important

neilweinstock, 2020-11-12:
Sorry, I was confusing the two issues (spill holes and shroud line length).

I do agree that more accurate descent rate calculation would be very valuable, but only if there is sufficient info out there to create a reasonable analytical model of it.

hcraigmiller, 2020-11-12:
Parachute design reference.
[https://apps.dtic.mil/dtic/tr/fulltext/u2/a247666.pdf](url)