openrocket/openrocket #819: Fins shape and CP position

apxaerospace, 2020-11-11:
Hello to everyone!
I find that making 1 fin set generate a CP position that is completely off if I use 2 fin set. I try to explain a bit more accurate :)
If I create 2 fin set with the same size and position of 1 fin set, I obtain a different CP position (same location, same size, no rotation between the sets). Just try to split a square fin in 2 smaller squares and place close together. The CP position in completely off. 
Anyone noticed it?
Thanks!

hcraigmiller, 2020-11-11:
OpenRocket assumes that when you create 2 different sets of fins that you intended to apply every aerodynamic aspect of "separate" fin sets. Meaning, the fluid dynamics associated with the airflow over each separate fin set apply; **OpenRocket does not recognize "butted" fins**.  In other words, _regardless of how many pieces a fin may be constructed of_, in OpenRocket _that fin is one fin_.

The easiest way to see this is to create the two abutted fin sets, then rotate one of the fin sets. The CP remains constant, meaning that each fin set is treated aerodynamically as completely separate and distinct, even when abutting another fin set.

Recommend closing this issue.

apxaerospace, 2020-11-16:
Hello Craig, so based on "regardless of how many pieces a fin may be constructed of, in OpenRocket that fin is one fin" the CP should remain the same with 1 big set and with 2 smaller (but equale to the first one) but based on your phrase, the openrocket behaviour is not the same. I'm not an aerodynamic specialist, but I don't know which CP position is the right one. 
[image]
[image]

hcraigmiller, 2020-11-16:
Forgive me for my prior poor explanation.  _Because_ OpenRocket assumes that when you create 2 different sets of fins that you intended to apply every aerodynamic aspect of "separate" fin sets..., regardless of how many pieces a fin may _actually_ be constructed of, in OpenRocket that fin _must be entered as_ one fin.

So, to your first "split fin" example, assuming that each fin has been entered as a single fin, the CP would be correct. 

As to your second example, let me ask a question similar to your own, to demonstrate. Is the CP position of the design shown below correct?

[image]
The answer, there is insufficient information because a 3D view has been presented, not the side view which shows the actual components.

So let's look at side view possibilities for the examples you presented. First, the split fin.

[image]

Each fin is drawn as a single fin, so the CP will be correct.

 Next, in 3D view, it looks like a single fin. But, is it? **Is it _drawn_ as a single fin**, with the **correct CP**...

[image]

Or, **is it _drawn_ as two fins**, with an **incorrect CP**...

[image]

**Incorrect CP** _because_, regardless of how many pieces a fin may _actually_ be constructed of, in OpenRocket that fin **_must be entered as one fin_**.

I hope the foregoing answers your question.

apxaerospace, 2020-11-17:
Hi Craig, 
thanks again for your reply. So becasically we can say:
- Separated fins set always shown CORRECT CP position
- (abutted fins) must be designed as one fin only
Aerodinamically speacking, 2 fin set is better than 1 fin set for CP purpose. No matter about the total area based on that drawing. Right? If I put 4 fin set with the same total area, I will get a better CP position compared to 1 fin set with the same area.

hcraigmiller, 2020-11-17:
When you split a fin and move part of it forward, it creates a different airflow pattern over both fins with a CP position outcome that can be calculated by OpenRocket.  Because OpenRocket does not recognize abutted fins as a single fin, abutted fin CP positions may not be accurately calculated. That is all I am saying.

closed 2020-11-17 by apxaerospace