openrocket/openrocket #825: Update checker still checking SourceForge

jimmiedave, 2020-11-18:
Noted that on startup, OpenRocket is still checking SourceForge for updates. The update URL is apparently specified in net.sf.openrocket/communication (not a Java coder; please forgive if this path to the trouble isn't well-specified).

I'm assuming it should be checking github.

Also noting that "sourceforge" appears a few times in the code elsewhere, and many times in "web" documentation that's in the repo.

```
public abstract class Communicator {

	protected static final String BUG_REPORT_URL;

	protected static final String UPDATE_INFO_URL;
	
	static {
		String url;
		url = System.getProperty("openrocket.debug.bugurl");
		if (url == null)
			url = "http://openrocket.sourceforge.net/actions/reportbug";
		BUG_REPORT_URL = url;
		
		url = System.getProperty("openrocket.debug.updateurl");
		if (url == null)
			url = "http://openrocket.sourceforge.net/actions/updates";
		UPDATE_INFO_URL = url;
	}

```

wolsen, 2020-12-03:
I think this all needs a bit of rework. The update url returns a 404 and I'm not sure of the infrastructure that's serving this. Will need to look at it closer.

neilweinstock, 2021-05-10:
Suggestion: remove it completely until it works properly with updated mechanism.

neilbalch, 2021-05-11:
I'm interested in potentially working on this one. Is there an analogous set of resources we'd like it to point people to on GitHub?

I presume that the bug tracker should link to the GitHub issues tracker and the updater should check the GitHub releases page...

closed 2022-01-28 by JoePfeiffer

jimmiedave, 2022-01-28:
> Closed #825 via #1082.

This is where GitHub needs a "like" button. Awesome!

DC

***@***.***>

PR #1082 (2022-01-28): [fixes #825] Rewrite Software Updater
This issue fixes #825 by rewriting OR's software update checker to use GitHub's release page.

**[Practical implementation]**
There are 4 different situations when checking for updates:

1. Something went wrong (e.g. bad internet connection) => a warning message will be shown:
[image]

2. Nothing went wrong and the current build version is the latest version, then the user will get a popup like this:
[image]

3. The current version is "newer" than the latest official release. This can be in the case of a beta version like we have now (the unstable build ve [... cut]
commit 201ab91e (2022-01-28): [fixes #825] Clean up translations no updates available
Move them to a dedicated section for the software update checker + give them a logical name.
commit 23af62d8 (2022-01-28): [fixes #825] Change update URLs
commit 3285a326 (2022-01-28): [fixes #825] Include commonmark library JAR-file
Sorry, forgot to do this 2 commits ago...