openrocket/openrocket #826: Upgrading to Java 15 causes unit test failures

JoePfeiffer, 2020-11-18:
Updating to either
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 15.0.1+9, mixed mode, sharing)
or
Eclipse OpenJ9 VM AdoptOpenJDK (build openj9-0.23.0, JRE 15 Linux amd64-64-Bit Compressed References 20201022_81 (JIT enabled, AOT enabled)
under Linux results in the following two unit test failures:

TestScriptingUtil	testGetLanguages	Failure	N/A
junit.framework.AssertionFailedError
at net.sf.openrocket.simulation.extension.impl.TestScriptingUtil.testGetLanguages(TestScriptingUtil.java:41)
at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:64)
at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)

TestScriptingUtil	testGetLanguage	Failure	expected:<JavaScript> but was:<null>
junit.framework.AssertionFailedError: expected:<JavaScript> but was:<null>
at net.sf.openrocket.simulation.extension.impl.TestScriptingUtil.testGetLanguage(TestScriptingUtil.java:32)
at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:64)
at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)

I'm not getting the unit test failures with Java 14.

hcraigmiller, 2022-02-12:
Where we will now be looking at JAVA 17 LTS compatibility, I recommend closing this issue.

JoePfeiffer, 2022-02-12:
Unless the problem still exists under 17, of course. Needs to be checked.

SiboVG, 2022-02-12:
This issue still exists under Java 17

PR #1507 (2022-08-02): Resolving issue #1270, unexpected exception with simulation scripting
This PR fixes #1270, fixes #308, fixes #1108 and fixes #826.

As the Nashorn scripting engine is deprecated, as part of the resolution this replaces Nashorn with the Graal.Js scripting engine.  This includes the addition of the following dependencies

graal-sdk-22.1.0.1.jar
icu4j-71.1.jar
js-22.1.0.1.jar
js-scriptengine-22.1.0.1.jar
truffle-api-22.1.0.1.jar
closed 2022-08-02 by SiboVG