openrocket/openrocket #827: Launch failure in macOS Big Sur

kenbiba, 2020-11-21:
Just won't launch.

JoePfeiffer, 2020-11-22:
Assuming you're trying to launch 15.03, it requires an antique version of Java.  You can find a prepacked installer for MacOS that includes the needed version at https://github.com/openrocket/openrocket/releases/download/release-15.03/OpenRocket-15.03.dmg

kenbiba, 2020-11-22:
Sadly ... it does not launch.

JoePfeiffer, 2020-11-22:
That's with the dmg, not the jar file? We'll see if anyone else has any ideas....

kenbiba, 2020-11-22:
Sadly yes.   Just tried again.   At first it complained about Java.  Now it is just silent and does nothing.

Perhaps you can open this issue again.

jimmiedave, 2020-11-22:
It's possible that MacOS gatekeeper is preventing unsigned code from running - but there's usually a dialog that comes along with that.

Can you describe what the install process was like as you installed from the DMG - you probably would've gotten a dialog that says "this file was downloaded from the web - do you want to open it?"

Then there may have been other administrator dialogs, or other complaints from the system. It'd be helpful to know what they were.

Are you an administrator on the system? (if you're the only person who's ever used it, and it's not managed by a business, yes, you probably are).

jimmiedave, 2020-11-22:
You may also be able to go into Security and Privacy (in System Preferences) and find a place to set an unsigned application to run - but I don't see such a place on my Catalina system.

kenbiba, 2020-11-22:
Hmmmm .... fixed it.

I removed the old one first.  And emptied trash.  Then reinstalled from the source.

This time it explicitly asked for authentication - requiring me to option-click on the first open and allow it to open.

It launched.   For the moment it seems OK.

Thanks for responding.

JoePfeiffer, 2020-11-23:
Don't know what went wrong the first time... but glad it's OK now.

closed 2020-11-23 by JoePfeiffer