openrocket/openrocket #835: New feature request: Ability to set the freeform fin shape grid size to values other than .5 inches

hcraigmiller, 2020-12-04:
Although a grid size of .5" is acceptable for medium and large rockets, many rocketeers design and fly small, mini, and even micro rockets. When you have a body tube that is .5" or less in diameter, the fins are small as well, and a grid size of .5" may only hit a fin relatively few times, giving very little help to the designer. See the example attached below.

[image]

It would be extremely helpful to allow the user to select the number of lines per inch (small squares) and the number of squares within the the heavy index (large squares); many designers use fractional marks rather than 1/10 marks. for the heavy index lines.

neilweinstock, 2020-12-12:
Preferable would be the smart grid we had in 15.03.  This is not something that should need another user setting.

neilweinstock, 2021-05-10:
The smart grid has been back for a while.  I think we can close this one?

hcraigmiller, 2021-05-10:
I reviewed the function of the auto grid sizing feature and verified that the size of the grid automatically changes to a value smaller than .5" when designing micro rockets. In light of the need to reduce issues to hasten the next release, I believe that the current unstable functionality is adequate for most users.

closed 2021-05-10 by hcraigmiller