openrocket/openrocket #845: Motor configuration/simulation shown in reverse order

hcraigmiller, 2020-12-10:
Using the Simple model rocket design, delete all of the motor configuration, then add 4 new configurations. Name each configuration with motors from A to D. Then, select the Flight simulations tab. The simulations are in reverse order to the motor configurations.  Version 15.03 is in the correct order.

[image]

And, when you add an additional motor, in the current unstable it adds it to the front of the list, where 5.03 adds it to the back of the list.

neilweinstock, 2020-12-22:
IMHO the UI for both these tabs is in sore need of a makeover, in many areas.  I agree the order of the listings is borked.

jimmiedave, 2021-01-31:
Getting the same results here. Also (if someone is making the UI over), there needs to be 

1. Some sort of persistent re-arrangement of the motor choices so one can arrange by (f.e.) letter, or in order of total impulse, or alpha by designation, or _something_, so that one can make sense of it.

2. A way to export the configuration selection: Chances are that I have one set of reloadable motors and one superset of things that fit them. Wouldn't it be good to set that all up once, export it, load it into each rocket model in turn and delete those I wouldn't use, vs. creating everything from scratch for each model?

jimmiedave, 2021-02-23:
Poking around in the source, looking at FlightConfigurableParameterSet, it looks like motors are loaded and indexed with a hash table, and with (I _think_) iteration on these hashes (FlightConfigurationId), they can come out in any old order (compared to the human way of thinking: alphanumeric, or even better: the way they were in the input file). Ditto for inserting into the FlightConfigurations.

Unfortunately, I don't have the Java chops and background on OR code to fix or modify. Seems like one could add an integer column or property to each FlightConfiguration as it's ingested and iterate on that, add the next one at highestinteger+1. Iterate again on them to write them out in order.

Looks like there are a couple of sort methods previous coders have tried, but actually, sorting is **not** what OR should be doing: it should honor the organization of the FlightConfigurations in the input file, and write them back to file respecting the original order - which it does not in unstable, at present. 15.03 handles the sort order as it finds it.

teyrana, 2021-02-24:
The behavior you're describing ... it sounds like you just want to be able to sort the displayed table, above.
Ff you're feeling ambitious... adding a sorter to the JTable: https://stackoverflow.com/questions/28823670/how-to-sort-jtable-in-shortest-way  seems like a reasonable approach.

-----
> ... it looks like motors are loaded and indexed with a hash table, and with (I think) iteration on these hashes
> (FlightConfigurationId), they can come out in any old order (compared to the " [displayed label] 

This is correct -- because a user can change the motors in a given configuration, the _label_ can change.   So the only consistent thing to track on a FlightConfiguration is the id, which is, essentially a randomly generated number.  Keep in mind, this ties in to separation and deployment configurations, too -- They have to, because they depend on the motor configuration.
Unfortunately, there IS NO natural ordering.  And the hashtable _explicitly is not ordered.  According to spec: "It makes no guarantees as to the iteration order of the set; in particular, it does not guarantee that the order will remain constant over time."

> .... sorting is not what OR should be doing: ....

Hmmmm, keep in mind it's all OR.  All the way down. The file format is openrocket. The file savers are open rocket.  The file loaders are open rocket.   Your only choice you get is _which_  _part_ of OR should sort that data.  

> .... it should honor the organization of the FlightConfigurations in the input file, and write them back to file respecting the original order

Let's just take a look at that input file format:
```
    <motorconfiguration configid="b13e8ca2-f54d-4ca8-9b23-06c6c9eb237a"/>
    <motorconfiguration configid="da326836-0959-4c94-bcd5-49dee07235a4"/>
    <motorconfiguration configid="e1dc7488-77c0-403a-b54f-c01db07aba2f" default="true"/>
    <motorconfiguration configid="05896b5a-71a7-48ac-b7ae-eadeb267975b"/>
    <motorconfiguration configid="5d9a2765-395d-4d9c-9678-38eed0988c72"/>
```
.... That's from the previous stable version. Those 'configid' values are the keys into the hash table of FlightConfigurableParameterSet.

jimmiedave, 2021-02-24:
Clearly a hash table does not sort in a predictable order. It couldn't do the things that are expected of hashes (SHA-256, etc.) if it did.

However, it's quite common to add a column to a database table, or a property to an object to assist with sorting for human consumption. 

Hash tables have no natural order, but displays of information should. (If this is not obvious, suggest you see the "[5 hat racks](https://www.amazon.com/Hats-Design-quarterly-Richard-Wurman/dp/B00072ALZ0)" article by Richard Saul Wurman, if you can find a copy.

I did look at sorting earlier today, but that's really not what I'm suggesting. I'm instead suggesting that OpenRocket do two things, and that not doing those two things is unexpected and incorrect behavior:

1. Respect the sort order in the OpenRocket file, preserve it, save it, or if a new rocket model, respect and save the order in the motors and simulation panels, and respect the saved order when loading the file. 15.03 does so. It should work like that.
2. When adding a new configuration, add it to the end of the list of configs by default. (One might also add a feature where selecting a config first adds a row above that config (so one could add at the head), but that would be a nice-to-have). 15.03 added predictably. Unstable does not do so, but should.

15.03 respects the order from the configuration screen  when saving a file. I've added "custom" typed names rather than used the defaults here. The names start with an ordinal, but this doesn't have anything to do with the sorting. It just works, even with default names:

[image]

15.03 loads this file up and respects the saved order of configurations. A compile of the latest I can pull down from github mangles the order completely, and will save the file in mangled order.

[image]

That hashes are used in 15.03 is beyond question. But I believe the ordering is supplied by the ordinal position of the motorconfigurations in the loaded file, not by sorting by hash.

jimmiedave, 2021-02-24:
I'll further point out that when someone has a crap-ton of motor options, it gets really hard to manage them when OpenRocket unstable scrambles them up. The following excerpt of my motor configs (not all of which are viable with this model) is manageable under 15.03. Definitely not on current unstable because it makes a mess of things when it loads up the file.

[image]

wolsen, 2021-06-05:
Honestly, I think the motors and configuration tab users are expecting that it is sorted in the order they were added in (since you can't sort by column name).  This is highlighted by @jimmiedave's latest comment here.

I think this is a simple change - instead of a HashMap which has inconsistent ordering, we can use a LinkedHashMap which preserves order by insertion. This way the configurations are returned as they are inserted - either manually by the user or what is loaded from the saved rocket document.

PR #936 (2021-06-12): FlightConfigurableParameterSet Predictable Order
Change to use a LinkedHashMap instead of a HashMap in the
FlightConfigurableParameterSet. This ensures that the ordering is
consistent with the way the user added the flight configurations.

This will also be true for when reading the flight configurations from
a file. Additionally, to preserve the ordering, do not sort the
FlightConfigurationIds in getIds().

Fixes #845
commit 965311ff (2021-06-12): FlightConfigurableParameterSet Predictable Orders

Change to use a LinkedHashMap instead of a HashMap in the
FlightConfigurableParameterSet. This ensures that the ordering is
consistent with the way the user added the flight configurations.

This will also be true for when reading the flight configurations from
a file. Additionally, to preserve the ordering, do not sort the
FlightConfigurationIds in getIds().

Fixes #845
closed 2021-06-12 by JoePfeiffer