openrocket/openrocket #849: Add hardware components to designs

JoePfeiffer, 2020-12-14:
It would be nice to be able to add hardware components to designs -- screws, eye bolts, washers, nuts, etc, to assist in developing electronics bays.

neilweinstock, 2020-12-14:
Curious, are you looking for them to be represented in the component tree, or visually in the 2D or 3D renderings?  Right now, "Mass Component" offers a bunch of subcategories that are rendered differently in the component tree (I only discovered this recently).  It would be easy to add additional options to this list.  However, all are rendered as the same blob in the side and 3D views.

What exactly did you have in mind?

neilweinstock, 2020-12-14:
Slight correction: different mass component types are actually shown slightly differently in side view (although kind of ugly IMHO), but the same generic blob in 3D views.

JoePfeiffer, 2020-12-14:
And I discovered the "Mass type" combobox just now, while looking at the "Mass Component Configuration" after seeing your response!  After looking at it, I guess my suggestion becomes "add more to the list" (I'm thinking specifically of fasteners -- screws, nuts, washers, eye bolts), and "better renderings".

hcraigmiller, 2021-03-29:
By user K' Tesh ktesh@hotmail.com:

"I'm working on sims that have rail buttons with screws and blind nuts.  Others that have plastic rivets, and most recently, kits that include sleds. I know that rail buttons have been added, but what about the screws and nuts that hold them in place?  Lots of people use plastic rivets for shear pins etc.  and lots of them have sleds.  I realize that I can use PBTs and mass objects to sim these, but it'd be nice if they don't need to have visual elements deleted to give accurate flight performance.  So, I'd like to propose the addition for these objects."